boltgolt/howdy #123: howdy didn't work with gui Apps

jamy3531, 2019-01-13:
_Please describe the issue in as much detail as possible, including any errors and traces_

Howdy works fine in terminal every time, but failed in Gui apps, Gparted, screen unlock for example.
Howdy worked very well for all applications before last update(the latest version).

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Deepin Linux

Howdy version: 2.5.0

jamy3531, 2019-01-13:
The camera light blinks but verification can not passed for gui applications

boltgolt, 2019-01-13:
Could you try running `sudo howdy config` and setting no_confirmation to `true`?

jamy3531, 2019-01-14:
@boltgolt I did so, but it made no difference

boltgolt, 2019-01-16:
Are there any python errors in `/var/log/auth.log`?

jamy3531, 2019-01-18:
There's no such file in my system.

Strangely， everything’s ok again after my deepin system update.

boltgolt, 2019-01-18:
`/var/log/auth.log` is used to log authentication errors on all debian distro's i've seen so far, so that's weird. Good to hear it's somehow fixed though, i'll mark it as closed for now. Don't be afraid to reopen it if you have the same issues again.

closed 2019-01-18 by boltgolt

boulchci, 2020-10-31:
I open the issue because the same problem happened to me. When I open the log file the only error is `[HOWDY]: Failure, timeout reached`.
Is there any solution ?