boltgolt/howdy #127: Outdated Package on Fedora COPR

Superdanby, 2019-01-20:
I just saw this amazing project. I would like to check out the latest release on Fedora, but the package on COPR is 2.4.0 instead of 2.5.0.

boltgolt, 2019-01-22:
@luyatshimbalanga is the creator of the original COPR package, and as i'm too unfamiliar with fedora he's working on the 2.5.0 build. A lot has changed in the way Howdy is installed in 2.5.0 so it's not just updating the version number.

Superdanby, 2019-01-23:
Looking forward to it. Thanks!

luyatshimbalanga, 2019-01-23:
howdy 2.5.0 is now available in the repository. Unfortunaly, new dependencies aren't available on Fedora yet to start the application and keep on failing to built. Investigating...

boltgolt, 2019-01-24:
Either way thanks a lot for putting in the effort, is it ok if i add you as the fedora maintainer in the README?

luyatshimbalanga, 2019-01-25:
Sure.

mattyokan, 2019-02-02:
Hey @luyatshimbalanga, this issue might not be the best place to ask for help (I apologize), but I'm looking to install your COPR's howdy package, and I'm running into an error when installing:

```  - nothing provides python3dist(pyv4l2) needed by howdy-2.5.0-1.fc29.x86_64
```
Is this a known issue with your repo or is this something stupid on my part? Thank you for maintaining this for Fedora, I can't wait to be able to use this project on my laptop.

Thanks!

luyatshimbalanga, 2019-02-03:
Hey @mattyokan,
yes  it is a known issue as pyv4l2 refused to build for some reasons. I haven't got a chance to investigate.

gstanden, 2019-02-03:
I ran into this same issue on Fedora 29 Workstation:

[orabuntu@localhost ~]$ sudo dnf install howdy
Last metadata expiration check: 0:14:21 ago on Sun 03 Feb 2019 06:33:06 AM CST.
Error: 
 Problem: conflicting requests
  - nothing provides python3dist(pyv4l2) needed by howdy-2.5.0-1.fc29.x86_64''

I will close my issue as it is a duplicate of this one.

gstanden, 2019-02-03:
I also tried all of this, just hacking away at it ... still no luck ... HTH

https://bugzilla.redhat.com/show_bug.cgi?id=1651317

Superdanby, 2019-02-03:
@gstanden To my experience, we better not mess with system python configuration.

gstanden, 2019-02-03:
Heh thanks good point. This is just a scratch fedora 29 instance and I'm just swinging away at it no idea really what any of this is doing.  Thanks for posting that note to warn others who are working on "mission critical" fedora deployments.  Putting coins into the slot machine, hoping to hit ...

luyatshimbalanga, 2019-02-08:
An update, pyv2l2 successfully built for Fedora 28. Unfortunately, due to the change within gcc on variable starting from 8.2, the built failed on Fedora 29 and higher. [See the report](https://github.com/duanhongyi/pyv4l2/issues/3)

closed 2019-03-14 by boltgolt

iamzubin, 2019-03-14:
hey @boltgolt, how should i install it in fodora 29?
is there any way i build it from source?

boltgolt, 2019-03-26:
Probably, but the [issue](https://github.com/duanhongyi/pyv4l2/issues/3) on pyv4l2 hasn't exactly gotten a lot of attention

luyatshimbalanga, 2019-03-27:
@boltgolt @iamzubin 
Updated version is up using v4l2 rather than pyv4l2.