boltgolt/howdy #128: Pop!_OS Install on Thinkpad P1, "howdy test" is empty

eduncan911, 2019-01-20:
sudo howdy add

Does not detect face.

    sudo howdy test

Open a window that is blank - just the cream-white background of my theme. No green noise pixelated info, nothing.

However, the console is full of these:

````
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x3800010
````

I've tried purging and installing Howdy multiple times. Each time when it asks me if my IR lights are on, they are on - and I select Yes. I've tried answering No and let it cycle the rest of the cameras - however, none of them turn on the IR lights.

When I do select "Yes" to the IR lights on, after install and during `sudo howdy test`, it does show my IR lights. But the popup window is blank.

I've tried editing `sudo howdy config` to change the video manually to one of the others, without success.

----

I've searched for similar issues already, and my issue has not been reported yet.

Yes.  There is another P1 and X1 Extreme user reporting issues (same machine); however, my issue is different in that I do not get any image.

Linux distribution (if applicable):

Pop!_OS - Nvidia Edition (Ubuntu 18.10 with a skin job)

Howdy version:

```
$ sudo apt search howdy
Sorting... Done
Full Text Search... Done
howdy/cosmic,cosmic,now 2.5.0 all [installed]
  Howdy: Windows Hello style authentication for Linux.
```

boltgolt, 2019-01-22:
Did you try the solution in issue #60 already?

eduncan911, 2019-01-22:
Ah, I missed that in the search.  

I'm trying to re-install Howdy now...However, getting a python error after all the cmake build stuff:

```
>>> Cleaning up dlib

Temporary dlib files removed

>>> Installing OpenCV

Requirement already satisfied: opencv-python in /usr/local/lib/python3.6/dist-packages (4.0.0.21)
Requirement already satisfied: numpy>=1.11.3 in /usr/local/lib/python3.6/dist-packages (from opencv-python) (1.16.0)
Exception:
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/pip/_internal/cli/base_command.py", line 176, in main
    status = self.run(options, args)
  File "/usr/local/lib/python3.6/dist-packages/pip/_internal/commands/install.py", line 346, in run
    session=session, autobuilding=True
  File "/usr/local/lib/python3.6/dist-packages/pip/_internal/wheel.py", line 848, in build
    assert building_is_possible
AssertionError
Error while running last command
dpkg: error processing package howdy (--configure):
 installed howdy package post-installation script subprocess returned error exit status 1
Processing triggers for man-db (2.8.4-2) ...
Errors were encountered while processing:
 howdy
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

boltgolt, 2019-01-22:
The travis build also failed with that error, might be a temporary problem with numpy on pypi? It's breaking all installations at the moment though.

eduncan911, 2019-01-22:
If it helps, here are the packages that were updated as of today:
```
Log started: 2019-01-22  15:40:36
(Reading database ... 211149 files and directories currently installed.)
Preparing to unpack .../libapt-pkg5.0_1.7.0ubuntu0.1_amd64.deb ...
Unpacking libapt-pkg5.0:amd64 (1.7.0ubuntu0.1) over (1.7.0) ...
Setting up libapt-pkg5.0:amd64 (1.7.0ubuntu0.1) ...
(Reading database ... 211149 files and directories currently installed.)
Preparing to unpack .../libapt-inst2.0_1.7.0ubuntu0.1_amd64.deb ...
Unpacking libapt-inst2.0:amd64 (1.7.0ubuntu0.1) over (1.7.0) ...
Preparing to unpack .../apt_1.7.0ubuntu0.1_amd64.deb ...
Unpacking apt (1.7.0ubuntu0.1) over (1.7.0) ...
Setting up apt (1.7.0ubuntu0.1) ...
(Reading database ... 211149 files and directories currently installed.)
Preparing to unpack .../apt-utils_1.7.0ubuntu0.1_amd64.deb ...
Unpacking apt-utils (1.7.0ubuntu0.1) over (1.7.0) ...
Preparing to unpack .../poppler-utils_0.68.0-0ubuntu1.4_amd64.deb ...
Unpacking poppler-utils (0.68.0-0ubuntu1.4) over (0.68.0-0ubuntu1.3) ...
Preparing to unpack .../libtiff5_4.0.9-6ubuntu0.1_amd64.deb ...
Unpacking libtiff5:amd64 (4.0.9-6ubuntu0.1) over (4.0.9-6) ...
Preparing to unpack .../libpoppler-glib8_0.68.0-0ubuntu1.4_amd64.deb ...
Unpacking libpoppler-glib8:amd64 (0.68.0-0ubuntu1.4) over (0.68.0-0ubuntu1.3) ...
Preparing to unpack .../libpoppler79_0.68.0-0ubuntu1.4_amd64.deb ...
Unpacking libpoppler79:amd64 (0.68.0-0ubuntu1.4) over (0.68.0-0ubuntu1.3) ...
Setting up libapt-inst2.0:amd64 (1.7.0ubuntu0.1) ...
Setting up apt-utils (1.7.0ubuntu0.1) ...
Setting up libtiff5:amd64 (4.0.9-6ubuntu0.1) ...
Setting up libpoppler79:amd64 (0.68.0-0ubuntu1.4) ...
Processing triggers for libc-bin (2.28-0ubuntu1pop1) ...
Processing triggers for man-db (2.8.4-2) ...
Setting up libpoppler-glib8:amd64 (0.68.0-0ubuntu1.4) ...
Setting up poppler-utils (0.68.0-0ubuntu1.4) ...
Processing triggers for libc-bin (2.28-0ubuntu1pop1) ...
Log ended: 2019-01-22  15:40:44
```
Besides LibreOffice updates, I've installed a few other GNOME apps and tools.

boltgolt, 2019-01-25:
Looks like it's fixed. The ol' wait a few days and try again worked

eduncan911, 2019-02-07:
And back to our regularly scheduled program...

So yes, `sudo QT_X11_NO_MITSHM=1 howdy test` does indeed show the image (after I set the jpeg config option to true).

However, when running `sudo QT_X11_NO_MITSHM=1 howdy add` or `sudo howdy add`, it just sits there with the IR emitters on with me playing a game of no-blink with the camera for 30-60 seconds before it wins.

Is that normal?  should I stay starring at the camera for longer than a minute?

boltgolt, 2019-02-07:
It definitely should not, in normal circumstances it takes about 2 seconds to add a face model. If the add command never reaches a timeout it's stuck on dark frame skips. Because of the way the emitters flash, some frames will be unlit. Howdy checks if a frame is very dark and automatically skips it, but in your case it seems like it's skipping all frames.

You could up the `dark_threshold` value to allow more dark frames to be parsed. For debugging purposes the test command shows if a frame would be skipped by printing "dark frame" in the top right.

You can safely run the add command without `QT_X11_NO_MITSHM=1` btw. Only the test commands spawns a qt window and thus needs that fix.

boltgolt, 2019-02-07:
Something that should be fixed in 2.6.0 is that the test command "dark frame" text does not use the `dark_threshold` config value and in stead always uses a static setting of 50.

eduncan911, 2019-02-07:
It worked!  I was able to add, and then use it to login.

Question: What's the valid range for `dark_threshold`?

boltgolt, 2019-02-07:
Great to hear! It's a percentage, so values from 0 to 100 work.