boltgolt/howdy #130: Why was this changed? It broke for me.

timwelch, 2019-01-21:
Line 11 in pyv4l2_reader.pl should be:

        from pyv4l2.frame import Frame

Instead of this, as v4l2 does not contain a 'frame' object:

        from v4l2.frame import Frame

----

Howdy version: latest

timwelch, 2019-01-21:
also, this line breaks for me, I have to remove the 'from recorders' piece to make it work.

`from recorders import v4l2`

timwelch, 2019-01-21:
```
>>> from v4l2.frame import Frame
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'v4l2.frame'; 'v4l2' is not a package
>>> 
```

boltgolt, 2019-01-22:
For some reason i [renamed it](https://github.com/boltgolt/howdy/commit/3c8db931b961f4e5f1ff18c38e47bbe3c5c1d41e#diff-ac54ae25ee8c75198508ef38189b8676) while moving the recorders to a new folder (no clue why). I'll fix it in the next release, sorry.

boltgolt, 2019-03-29:
Fixed the typo in 2.5.1

closed 2019-03-29 by boltgolt