boltgolt/howdy #132: Howdy entirely broken after recent update

Ethiarpus, 2019-01-22:
_Please describe the issue in as much detail as possible, including any errors and traces_

Just ran apt update && apt upgrade, and howdy immediately stopped working. I don't think OpenCV was upgraded this time around.
Happy to post any additional infomation, but here's the traceback thrown to console:

```
Traceback (most recent call last):
  File "/lib/security/howdy/compare.py", line 210, in <module>
    face_locations = face_detector(gsframe, 1)
RuntimeError: Unsupported image type, must be 8bit gray or RGB image.
Unknown error: 1 
```

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable):

Mint 18.3 sylvia
x86_64 Linux 4.15.0-42-generic

Howdy version:
2.5.0

boltgolt, 2019-01-23:
I'll keep it open for now as you probably won't be the last to encounter this. In 2.6.0 i'll add a catch for the runtime error.

Ethiarpus, 2019-01-24:
@boltgolt 
Update; output from `sudo howdy test`: 

```
Opening a window with a test feed

Press ctrl+C in this terminal to quit
Click on the image to enable or disable slow mode

Traceback (most recent call last):
  File "/usr/local/bin/howdy", line 101, in <module>
    import cli.test
  File "/lib/security/howdy/cli/test.py", line 159, in <module>
    face_locations = face_detector(frame, 1) # upsample 1 time
RuntimeError: Unsupported image type, must be 8bit gray or RGB image.
```

boltgolt, 2019-01-24:
Same problem, with about the same fix. In `/lib/security/howdy/cli/test.py` remove line 109:

```python3
frame = cv2.cvtColor(frame, cv2.COLOR_BGR2GRAY)
```

boltgolt, 2019-02-02:
The XPSs have some really funky GREY8 encoded cameras that until fairly recently did not have a linux driver. A new driver was merged in march last year (see [this](https://github.com/boltgolt/howdy/issues/85#issuecomment-432444371) comment) so only the newest systems will come with the right kernel by default.

ntrepid8, 2019-02-02:
I think version 4.19 is the first Linux kernel with GREY8 support.

boltgolt, 2019-03-29:
Should be fixed by 2.5.1, let me know if it isn't

closed 2019-03-29 by boltgolt