boltgolt/howdy #133: Sub-process /usr/bin/dpkg returned an error code (1)

scampson, 2019-01-22:
Using a HP Spectre X360 2017 model running Ubuntu 18.10. I've used Howdy previously, but never managed to get the IR emitters to work. Finally got round to installing Howdy again and received the below error towards the end of the install:
"Exception:
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/pip/_internal/cli/base_command.py", line 176, in main
    status = self.run(options, args)
  File "/usr/local/lib/python3.6/dist-packages/pip/_internal/commands/install.py", line 346, in run
    session=session, autobuilding=True
  File "/usr/local/lib/python3.6/dist-packages/pip/_internal/wheel.py", line 848, in build
    assert building_is_possible
AssertionError
Error while running last command
dpkg: error processing package howdy (--configure):
 installed howdy package post-installation script subprocess returned error exit status 1
Processing triggers for libc-bin (2.28-0ubuntu1) ...
Errors were encountered while processing:
 howdy
E: Sub-process /usr/bin/dpkg returned an error code (1)"

Not sure what's happened, this is a pretty new Linux install. Couldn't find anything on this error in my searches, but if it has come up before apologies in advanced.

boltgolt, 2019-01-22:
It was reported in #128, but not much before you. It's an dependency that spontaneously started to fail installing in the last hour or so. I'll give it some time to also spontaneously fix itself, if it doesn't i'll pin the dependency to a lower version.

boltgolt, 2019-01-25:
Looks like travis is testing normally again, the numpy issue is fixed

closed 2019-01-25 by boltgolt