boltgolt/howdy #134: Howdy add

Naragorn, 2019-01-23:
I have the Dell XPS 13 9370, howdy test shows me my face in a greyscale image and recognizes my face.
However after running howdy add I get the following error:

Please look straight into the camera
Traceback (most recent call last):
  File "/usr/bin/howdy", line 90, in <module>
    import cli.add
  File "/usr/lib/security/howdy/cli/add.py", line 123, in <module>
    enc = face_recognition.face_encodings(frame)
  File "/usr/lib/python3.7/site-packages/face_recognition/api.py", line 210, in face_encodings
    return [np.array(face_encoder.compute_face_descriptor(face_image, raw_landmark_set, num_jitters)) for raw_landmark_set in raw_landmarks]
  File "/usr/lib/python3.7/site-packages/face_recognition/api.py", line 210, in <listcomp>
    return [np.array(face_encoder.compute_face_descriptor(face_image, raw_landmark_set, num_jitters)) for raw_landmark_set in raw_landmarks]
TypeError: compute_face_descriptor(): incompatible function arguments. The following argument types are supported:
    1. (self: dlib.face_recognition_model_v1, img: numpy.ndarray[(rows,cols,3),uint8], face: dlib.full_object_detection, num_jitters: int=0, padding: float=0.25) -> dlib.vector
    2. (self: dlib.face_recognition_model_v1, img: numpy.ndarray[(rows,cols,3),uint8], faces: dlib.full_object_detections, num_jitters: int=0, padding: float=0.25) -> dlib.vectors
    3. (self: dlib.face_recognition_model_v1, batch_img: List[numpy.ndarray[(rows,cols,3),uint8]], batch_faces: List[dlib.full_object_detections], num_jitters: int=0, padding: float=0.25) -> dlib.vectorss

Invoked with: <dlib.face_recognition_model_v1 object at 0x7fdbf31fb260>, array([[ 35,  37,  35, ...,  36,  35,  37],
       [ 35,  34,  35, ...,  36,  36,  35],
       [ 35,  36,  35, ...,  37,  36,  38],
       ...,
       [135, 134, 136, ..., 105, 108, 108],
       [131, 134, 132, ..., 106, 103, 108],
       [134, 131, 133, ..., 103, 107, 106]], dtype=uint8), <dlib.full_object_detection object at 0x7fdbf31f1490>, 1

I am aware of issue #70 (https://github.com/boltgolt/howdy/issues/70), but the fix in there doesn't do anything for me.

Linux distribution (if applicable): Arch Linux

Howdy version: newest

boltgolt, 2019-01-23:
Could you try running `sudo howdy config` and changing the recording plugin to ffmpeg?

Naragorn, 2019-01-23:
howdy config returns this error:

Opening config.ini in the default editor
Traceback (most recent call last):
  File "/usr/bin/howdy", line 94, in <module>
    import cli.config
  File "/usr/lib/security/howdy/cli/config.py", line 12, in <module>
    subprocess.call(["/etc/alternatives/editor", os.path.dirname(os.path.realpath(__file__))  + "/../config.ini"])
  File "/usr/lib/python3.7/subprocess.py", line 323, in call
    with Popen(*popenargs, **kwargs) as p:
  File "/usr/lib/python3.7/subprocess.py", line 775, in __init__
    restore_signals, start_new_session)
  File "/usr/lib/python3.7/subprocess.py", line 1522, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: '/etc/alternatives/editor': '/etc/alternatives/editor'

Manually setting the config.ini

# Force the use of Motion JPEG when decoding frames, fixes issues with
# YUYV raw frame deconding
force_mjpeg = true

However still the same error...

boltgolt, 2019-01-24:
So arch is still on 2.3.1 while Ubuntu and Fedora are on 2.5.0. The newest versions of Howdy include a FFmpeg recorder that fixes your issue, but it isn't available on your disto yet. @joeknock90 was working on making a new arch build but it's not an easy upgrade.

sapjunior, 2019-01-26:
Have you ever try to edit the script file that I mentioned in https://github.com/boltgolt/howdy/issues/70#issuecomment-439123621?

Naragorn, 2019-01-28:
@sapjunior  Yes but that fix didn't do anything for me

Upgraded manually to 2.5.0 and it works like a charm with FFmpeg.
Someone had a comment in AUR to get the arch build of 2.5.0 (https://aur.archlinux.org/packages/howdy/)
The maintainer should upgrade the package...