boltgolt/howdy #136: Lenovo P72 Running Ubuntu 18.10

gstanden, 2019-01-27:
Hi I followed your instructions after reading your article [here](https://medium.com/@boltgolt/hacking-together-windows-hello-on-ubuntu-ecd97ff16430) and the install went fine on Ubuntu 18.10 running on a Lenovo P72 with no issues.  The IR cams came alive on the 3rd of the "Did your IR camera activate?" tests.  However, when I run "sudo howdy add" the IR camera blinks quickly just once and then howdy gives message "No face detected, aborting". Any suggestions how I can tweak/fix this? Thanks!

gstanden, 2019-01-27:
So I made some progress I think on this myself.  I ran a "sudo howdy test" and noted that the window that opened was blank with no image at all (no images, just the box with no content at all).  The error that was generated over and over was the [same as the one](https://github.com/P0cL4bs/WiFi-Pumpkin/issues/53) at this issue in another GitHub project.  So I followed the recipe there and added "QT_X11_NO_MITSHM=1" to /etc/environment and this seems to have "helped", at least it caused "sudo howdy test" to start working with no errors in the terminal and a VERY grainy almost now totally greyed out image appearing in the pop-up GUI box when running "sudo howdy test" but within a second or two the grained out foggy face just flips to almost a solid wall of grey and when running "sudo howdy add" it just hangs now, never gives the "no face detected, aborting" message.  One other side effect of adding this parameter to /etc/environment is that now the 4th test (and not the 3rd as before) for "did you IR emitter flash Y/N" is the one that triggers the IR emitters to flash.

gstanden, 2019-01-27:
I also tried installing "guvcview" as described [here](https://github.com/boltgolt/howdy/issues/91) in issue #91 but this did not enable "sudo howdy add" to succeed.  However, I did do some grep of dmesg and here is some output I found which looks like it might be relevant.

ubuntu@ubuntu-ThinkPad-P72:~$ sudo dmesg | grep "video"
[    3.473013] videodev: Linux video capture interface: v2.00
[    3.504020] uvcvideo: Found UVC 1.00 device Integrated Camera (04f2:b604)
[    3.552522] uvcvideo 1-8:1.0: Entity type for entity Realtek Extended Controls Unit was not initialized!
[    3.552524] uvcvideo 1-8:1.0: Entity type for entity Extension 4 was not initialized!
[    3.552525] uvcvideo 1-8:1.0: Entity type for entity Processing 2 was not initialized!
[    3.552526] uvcvideo 1-8:1.0: Entity type for entity Camera 1 was not initialized!
[    3.554931] uvcvideo: Found UVC 1.50 device Integrated IR Camera (04f2:b605)
[    3.558858] uvcvideo 1-12:1.0: Entity type for entity Microsoft Extended Controls Uni was not initialized!
[    3.558859] uvcvideo 1-12:1.0: Entity type for entity Realtek Extended Controls Unit was not initialized!
[    3.558861] uvcvideo 1-12:1.0: Entity type for entity Extension 4 was not initialized!
[    3.558862] uvcvideo 1-12:1.0: Entity type for entity Processing 2 was not initialized!
[    3.558863] uvcvideo 1-12:1.0: Entity type for entity Camera 1 was not initialized!
[    3.560328] usbcore: registered new interface driver uvcvideo
ubuntu@ubuntu-ThinkPad-P72:~$

gstanden, 2019-01-27:
Here's what I get from "tail -f" of the auth.log (the entries at 9:00:56 are the entries that popped in when I ran "sudo howdy add").

Jan 27 08:55:49 ubuntu-ThinkPad-P72 sudo: pam_unix(sudo:session): session closed for user root
**Jan 27 09:00:56 ubuntu-ThinkPad-P72 sudo:   ubuntu : TTY=pts/0 ; PWD=/home/ubuntu ; USER=root ; COMMAND=/usr/local/bin/howdy add
Jan 27 09:00:56 ubuntu-ThinkPad-P72 sudo: pam_unix(sudo:session): session opened for user root by (uid=0)
Jan 27 09:01:08 ubuntu-ThinkPad-P72 sudo: pam_unix(sudo:session): session closed for user root**

boltgolt, 2019-01-28:
Thanks for really trying to solve this issue on own! Sounds like you might be running in the wrong resolution, though. If you run `v4l2-ctl --list-formats-ext` you'll get a list of supported resolutions for all your cameras. Try using the ones listed for your IR camera as `frame_width` and `frame_height` in `sudo howdy config`.

gstanden, 2019-01-28:
Sorry total newb to IR emitters / IR cameras etc.  Which one is my IR ?  What do you recommend setting in howdy config ?

Here's the current settings from "sudo howdy config" for "frame_width" and "frame_height"

# Set the camera input profile to this width and height
# The largest profile will be used if set to -1
# Automatically ignored if not a valid profile
frame_width = -1
frame_height = -1

And here's the output from the command you provided - thanks!  Which one is the IR camera ?

ubuntu@ubuntu-ThinkPad-P72:~$ v4l2-ctl --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
	Index       : 0
	Type        : Video Capture
	Pixel Format: 'MJPG' (compressed)
	Name        : Motion-JPEG
		Size: Discrete 1280x720
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x180
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x240
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 352x288
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 424x240
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x360
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 848x480
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 960x540
			Interval: Discrete 0.033s (30.000 fps)

	Index       : 1
	Type        : Video Capture
	Pixel Format: 'YUYV'
	Name        : YUYV 4:2:2
		Size: Discrete 1280x720
			Interval: Discrete 0.100s (10.000 fps)
		Size: Discrete 320x180
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x240
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 352x288
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 424x240
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x360
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 848x480
			Interval: Discrete 0.050s (20.000 fps)
		Size: Discrete 960x540
			Interval: Discrete 0.067s (15.000 fps)

ubuntu@ubuntu-ThinkPad-P72:~$

gstanden, 2019-01-28:
Hey!  I changed it to 320 x 180 and the emitter flashed for a about 3 to 5 seconds (multiple flashes) and not just a single flash, and it captured the image!! Testing it now !

gstanden, 2019-01-28:
There's another problem.  Howdy works fine after I first configure it "for awhile."  But if I go away from the machine for say 20 minutes or so, when I come back, the IR isn't able to make the recognition match, and I get the following message (for example from a terminal sudo session attempt):

ubuntu@ubuntu-ThinkPad-P72:~$ sudo su -
Face detection timeout reached
[sudo] password for ubuntu: 
root@ubuntu-ThinkPad-P72:~# 

I can fix this by doing a "sudo howdy clear" and then doing a new "sudo howdy add" and then it works fine until I go away from the machine again for awhile, and then apparently for lack of a better term, the howdy/IR setup seems to "forget" my face and I have to go through the clear/add cycle again.

boltgolt, 2019-01-29:
So your keyring popup problem has been discussed in #39. It's not solvable without setting an empty keyring password.

The timeout is by design. You could up the timeout value in the config, but i wouldn't advice you to do so. While howdy is running you can't enter a password, so when IR isn't working due to things like sunlight, you can't log in until timeout.

gstanden, 2019-01-29:
Agreed on the keyring popup.  However, it's not the timeout that is a problem - it's the "forgetting" of the face recognition after say, 20 minutes.  What I'm trying now is different resolution settings in howdy config to see if that makes any difference.  For example, 424 x 280 also works for the Lenovo P72 (I was using 320 x 180) so perhaps when I find the highest resolution that will work, howdy will "remember" the IR credential image "permanently" ?

boltgolt, 2019-01-29:
Hm, might also be a confidence problem. Try adding an additional model when Howdy does not recognize you (instead of clearing) and maybe up the certainty value to 5 in the config. A higher resolution will also help if that's the case.

gstanden, 2019-01-30:
Thanks, it's working good now no issues with the following non-default settings in config:

certainty = 5
frame_width = 320
frame_height = 180

I did record two images named "ubuntu1" and "ubuntu2" as you suggested as well:

ubuntu@ubuntu-ThinkPad-P72:~$ sudo howdy list
Known face models for ubuntu:

	ID  Date                 Label
	0   2019-01-29 09:57:45  ubuntu1
	1   2019-01-29 09:57:58  ubuntu2

ubuntu@ubuntu-ThinkPad-P72:~$ 

but I think it's authenticating from ubuntu1 reliably (only takes a second or two and done) and the "forgetting" problem is gone now that I'm using "certainty = 5".

Thanks!!

gstanden, 2019-01-31:
Hi just one footnote fwiw.  It's all working great no "forgetting" anymore, except for one scenario, which is when the screen timeout occurs, and the desktop locks.  When I go to unlock the screen, hit enter, the IR emitters flash in the normal way for a few seconds, but the face read fails.  This doesn't happen on reboots, and it doesn't happen on terminal sudo sessions once the desktop is unlocked - both of those scenarios work perfect first time everytime.  But screen unlock isn't reading the face authentication and is failing.  Any ideas with that one?  Thanks!

boltgolt, 2019-02-01:
Sounds like one of the most reported problems, could you try setting `no_confirmation` to `true`? That sometimes throws a PamException: Conversation error with some screensavers,

gstanden, 2019-02-04:
Just a footnote to using this with the Lenovo P72 mobile workstation.   The P72 mouse handling and password screen handling at the GDM login screen is kind of herky jerky uneven when logging in to Ubuntu 18.10 on this machine.  Therefore, the configuration I am using with Howdy is VERY nice for handling login to the Ubuntu 18.10 desktop:  Howdy logs me in automatically at GDM login, and then the desktop now loaded presents a popup asking for the keyring password, which as noted above in this thread, has to be typed in.  This is a huge improvement on the login process in general for Ubuntu 18.10 boot/restart login on the Lenovo P72.  It gets you automatically past the quirky performance of GDM login page reliably and quickly, but still preserves a level of desktop login security.  Thanks!

closed 2019-02-04 by gstanden