boltgolt/howdy #137: Issue Installing on Surface Book "FileNotFoundError: [Errno 2] No such file or directory: '/dev/v4l/by-path'"

nriaziat, 2019-01-29:
I am getting this string of errors when attempting to install howdy on Ubuntu 16.04

```
Starting IR camera check...

Traceback (most recent call last):
  File "/var/lib/dpkg/tmp.ci/preinst", line 54, in <module>
    devices = os.listdir("/dev/v4l/by-path")
FileNotFoundError: [Errno 2] No such file or directory: '/dev/v4l/by-path'
dpkg: error processing archive /var/cache/apt/archives/howdy_2.5.0_all.deb (--unpack):
 subprocess new pre-installation script returned error exit status 1
Errors were encountered while processing:
 /var/cache/apt/archives/howdy_2.5.0_all.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)

```

boltgolt, 2019-01-29:
Looks like you don't have any v4l2 devices, could you check if you have any video devices active by running `ls /dev/video*`? Cloud you also check that v4l2 is installed (`dpkg -l v4l-utils`)?

nriaziat, 2019-01-29:
thanks for the comment! running the ls command said that there was no directory, and v4l2 is not installed as well. For reference, I am running on a Surface Book with custom kernel.

boltgolt, 2019-01-29:
So in short, video4linux is the default camera driver on basically all linux distributions. I've never seen it missing on a system before. Could your custom kernel have been compiled without video support?

nriaziat, 2019-01-29:
Had to check the kernel, but the JakeDay Surface kernel seems not to have camera support at this time. That must be the problem.

boltgolt, 2019-01-30:
You could try installing `v4l-utils` but that probably wouldn't fix anything. I don't think i can do much here, it's an kernel issue.

closed 2019-01-30 by boltgolt