boltgolt/howdy #140: Desktop Login Vs Sudo Auth

VanishingTacos, 2019-02-01:
Running Ubuntu 18.04 LTS

Logging in via the lock screen does not seem as effective as authenticating as root when you are already logged in. Sometimes the face detection will work when logging in but this seems to about less than 20 percent and when already logged in it works almost all the time. 

----

I've searched for similar issues already, and my issue has not been reported yet.
I do not believe this is a duplicate. 
Linux distribution (if applicable):
Running Ubuntu 18.04 LTS
Howdy version:
2.5.0

boltgolt, 2019-02-01:
Could you take a look at [this comment](https://github.com/boltgolt/howdy/issues/136#issuecomment-459772498) i just posted? Sometimes screensavers (lock screens) don't work well with Howdy for some reason.

VanishingTacos, 2019-02-02:
I kept I test with `no_confirmation` to `true` and `false` and I could not get it to working with it set to true. 

I think that `no_confirmation` does not effect this. I really think howdy is not super glasses friendly. Running `test` howdy seems to struggle really hard to detect my face. After adding some better model of me with glasses I was able to login just fine. I tested this a few times. 

My glasses really seems to throw it off a lot though.

boltgolt, 2019-02-02:
It might help to up the `certainty` config value to 5, which make Howdy accept more fuzy matches. I have glasses myself and with a separate model it works almost all of the time.

VanishingTacos, 2019-02-02:
Oh, wow that made a huge difference. Just did a quick test and had it work super fast on the lock screen and when doing root tasks. I will continue to test this though out the day.

VanishingTacos, 2019-02-03:
I had it set to 5 but I found out that it would unlock when I showed a picture of my face. I then lowered it to 4.5 and that seemed to have stopped it from happening. It is working almost every time now!

closed 2019-02-03 by VanishingTacos