boltgolt/howdy #141: Disable Howdy only for initial sing-in

wout, 2019-02-01:
I would like to be able to disable howdy for the initial log in process. Logging in with howdy does not unlock the keychain, which forces me to enter my password manually every time I need to make a ssh connection involving a ssh key. This can be avoided only when I do the initial login with password. But in order to do so, there is a significant time to wait until the password can be entered, while howdy is trying to get my face recognized.

boltgolt, 2019-02-01:
First of all, thanks for the suggestion! I agree with you that unlocking they keyring every time is a nuisance. There are ways to keep it [unlocked permanently](https://github.com/boltgolt/howdy/issues/56#issuecomment-405862123), but they are very insecure.

However, as explained [here](https://github.com/boltgolt/howdy/issues/85#issuecomment-432444371), Howdy does not know _why_ it is being called. The PAM call comes with no context. Detecting a first login would therefore have to rely on logging previous Howdy sessions or monitoring shutdowns. Both of which i would rather not do. Would getting #9 to work be a reasonable compromise?

wout, 2019-02-03:
Thanks for your prompt reply. And of course thanks for this excellent package, which I use a lot on a daily basis. :)

What's suggested in #9 seems like the ideal compromise. If I can start typing my password during face recognition, that would solve my issue completely. Then it's up to me to keep my face out of reach of the camera.

boltgolt, 2019-02-03:
I'll close this for now to keep this topic in #9

closed 2019-02-03 by boltgolt

Thalhammer, 2019-11-13:
You can work around this by temporarily disabling howdy during first login.
I am using elementary OS 5 which is based on Ubuntu (18.04 i believe) but it should work similar in other distros.
NOTE: "dominik" is my username and should get replaced with your own.
What I did was first create a file `/etc/sudoers.d/howdy` to allow executing howdy without a password.
`dominik ALL = (ALL) NOPASSWD: /usr/local/bin/howdy`
This is required for the following.

Create a task to run after login. I did not bother to look where to best place it as elementary os provides a nice gui for this. Make it execute the following command:
`/usr/bin/sudo /usr/local/bin/howdy disable 0`

Create a override in systemd to disable howdy before login. I picked "systemd-logind.service" as its a perfect fit, but it does not really matter which one you use as long as it is done before your login screen appears.
Create the file `/etc/systemd/system/systemd-logind.service.d/override.conf` and add the following:
`[Service]
ExecStartPre=+-/bin/bash -c "SUDO_USER=dominik /usr/local/bin/howdy disable 1"`

After a reboot it should work fine. Howdy is disabled until you login using your password which causes your keychain to be unlocked, and enabled for subsequent logins (like after standby or for sudo). This is the behaviour I wanted.

This could also be interesting for:
#234 

NOTE: All of this is not a proper solution but the best I came up with without error prone changes to howdy. It might break with updates to howdy or your OS, so don't rely on it.

eric1932, 2020-04-13:
I found that in howdy's configuration file, there is an option called "disabled." Maybe you can achieve the goal by changing this option. eg. disable howdy before shutdown, and enable it right after user's login

ryanerwin, 2020-06-30:
I would also definitely prefer to use Howdy as "screen unlock" rather than login... Mostly because of the keychain password issue.

@boltgolt said:
> Howdy does not know why it is being called. The PAM call comes with no context. Detecting a first login would therefore have to rely on logging previous Howdy sessions or monitoring shutdowns

From the context that Howdy is called under are you able to interact with the rest of the system?

If so, what if Howdy checks to see if a desktop environment exists via something like `loginctl` to see if the user already has a session (thus we're 99% sure it's a screen lock)?

lichaar, 2020-10-08:
This will be a really helpful enhancement, If I'm going to re-enter my password for keyring on every login, There is no point of using Howdy anymore :cry:

KairuiLiu, 2021-07-28:
For KDE, there is a sample way
only config `/etc/pam.d/kde`, do not add line in \*system\* and \*login\*
then it can only work on after you lock...

rokejulianlockhart, 2023-07-16:
@boltgolt, how would one request the developers of PAM to inform something like Howdy of the purpose of its invocation?

lynxnb, 2024-06-12:
@KairuiLiu 
> For KDE, there is a sample way only config `/etc/pam.d/kde`, do not add line in *system* and *login* then it can only work on after you lock...

Is there anything similar for GNOME?