boltgolt/howdy #146: face_recognition and dlib migration

acegallagher, 2019-02-06:
Hey there. I'm back. 

So I am loving howdy but I noticed that the import for face_recognition was taking forever, as you mentioned in your code comments. For fun I very quickly threaded the face_recognition import so that my auth took less time. Because the face_recognition import took ~3.0s, my auth times went down from 4.5s to 2.5s (for some reason the threaded instance of 'import face_recognition completes more quickly on my machine). 

Anyway. I came back here to submit a pull request and pass along my changes to improve the project and in doing that I find out that the git version has migrated to a dlib backend. Long story short. If for some reason you're satisfied with the dlib backend then you can ignore my feature request... but if you'd like I can happily provide the code that I've used for threading the import statements and making the rest of the code thread safe.

boltgolt, 2019-02-07:
First of all i want to thank you for working on improving Howdy and wanting to merge those changes back into the master! Open source projects like this one can only exist because of enthusiasts like you.

As you've seen yourself by now, the 2.5.0 release had a complete rework of the dlib import, most notably dropping face_recognition, adding threading and compiling with the most optimal settings on installation. These changes cut total recognition time by about half, just like your own changes did. 

However this does not mean that Howdy is as fast as it can possibly be. Importing still takes around 400ms (on my machine) and so does opening the camera. If your work can be helpful in the current version of Howdy, please do open an PR. Contributions are always welcome.

acegallagher, 2019-02-07:
Hey. Thanks for the prompt reply! Importing modules in my config takes only about 150ms (recent gen laptop core i5), while initializing the camera takes about 1.5 seconds o_o. The actual face recognition takes approximately 400ms. 

I imagine the difference between my code and your code is that some of the import calls I'm implicitly threading you're now importing separately of dlib. I would love to test the dlib migrated version but I can't get it to run, I get crashes trying to load all three video plugins. 

It's definitely possible that my version is faster because the entire face_recognition import will happen while the camera initializes but I think that might only be for a system like mine, where the camera starts up slow for some reason. 

If you're curious, I uploaded the threaded compare.py to my github. This of course will only drop in and work with Howdy 2.3, you'll have to test against the old version to see how it works. Here's the file, hope it finds you well: 

https://github.com/acegallagher/threaded_howdy/blob/master/howdy_compare_threaded.py

boltgolt, 2019-02-08:
I'll see what i can implement in Howdy, thanks!

closed 2019-02-08 by boltgolt