boltgolt/howdy #147: Unable to get test to work with IR camera on Lenovo P1

clegett, 2019-02-11:
Using the command `gst-launch-1.0 v4l2src device=/dev/video2 num-buffers=10 ! image/jpeg ! multifilesink location="frame-%02d.jpg"` works, yielding a number of 640x480 black and white images saved to disk. This also causes the IR/red LEDs to flash next to the camera. The same does not happen in cheese or in `sudo howdy test` when that device is selected. Cheese and `sudo howdy test` instead yield a very noisy green speckled image that vaguely resembles the anticipated scene and the IR/red LEDs stay off. 

I have tried setting the width and height in the config.ini file to both 640x480 and the cheese-detected 400x480 as well as leaving them set to -1.

boltgolt, 2019-02-11:
Could you try setting `force_mjpeg` to true?

clegett, 2019-02-11:
The behavior is the same with `force_mjpeg=true` and `force_mjpeg=false`

boltgolt, 2019-02-14:
Could you try to switch to the FFmpeg recorder? Not 100% sure what the problem is.

clegett, 2019-02-15:
How would I go about trying that? I don't see an appropriate setting in the config.ini file.

boltgolt, 2019-02-17:
If you're on the Debian or Fedora releases there should be a `recording_plugin` config option. If you're on Arch you will need to update to 2.5.0 which isn't officially updated yet. There are a few comments explaining how to update on the AUR page.

clegett, 2019-02-17:
Ah thanks for that. I upgraded, but once I do so and set the recorder to ffmpeg I get the following when running `sudo howdy test`:
```
$ sudo howdy test
Howdy has been configured to use a recorder which doesn't support the test command yet
Aborting
```
Simply testing with the default recorder yields the following image. The IR LEDs still do not illuminate.
[image]

The test and face detection works with /dev/video0 which is the normal webcam, but not the IR camera on /dev/video2.

boltgolt, 2019-02-20:
Do the emitters light up when running `sudo howdy add` with the FFmpeg recorder and the IR camera as device?

clegett, 2019-02-20:
No. It yields the following error:
```
$ sudo howdy add
Adding face model for the user clegett
Enter a label for this new model [Initial model] (max 24 characters): 
Traceback (most recent call last):
  File "/usr/bin/howdy", line 89, in <module>
    import cli.add
  File "/usr/lib/security/howdy/cli/add.py", line 129, in <module>
    video_capture.grab()
  File "/usr/lib/security/howdy/recorders/ffmpeg_reader.py", line 132, in grab
    self.read()
  File "/usr/lib/security/howdy/recorders/ffmpeg_reader.py", line 105, in read
    self.record(self.numframes)
  File "/usr/lib/security/howdy/recorders/ffmpeg_reader.py", line 89, in record
    .run(capture_stdout=True, quiet=True)
  File "/usr/lib/python3.7/site-packages/ffmpeg/_run.py", line 305, in run
    raise Error('ffmpeg', out, err)
ffmpeg._run.Error: ffmpeg error (see stderr output for detail)
```

mattyokan, 2019-02-21:
Are you running the latest BIOS version? I also have a P1 and I saw something about IR camera fixes in the most recent BIOS update, so that's worth checking.

clegett, 2019-02-21:
I'm on BIOS N2EET36W 1.18. Everything works fine under Windows and when using the gst-launch command above. It appears to be a problem with howdy somewhere, but I don't know how to track it down myself.

nomtat, 2019-03-23:
I'm also having this issue on the Thinkpad X1 Extreme, BiOS version 1.19. After installing howdy as per the instructions, the IR blasters don't turn on when I use the 'add' command.

clegett, 2019-03-24:
@nomtat This is still unresolved for me as well.

boltgolt, 2019-03-26:
Could one of you run `v4l2-ctl -d /dev/video2 --list-formats`? Make sure `/dev/video2` is your IR webcam

clegett, 2019-04-01:
Sorry for the delay in responding.
```
$ v4l2-ctl -d /dev/video2 --list-formats
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'MJPG' (Motion-JPEG, compressed)
	[1]: 'YUYV' (YUYV 4:2:2)
```

/dev/video2 is the IR webscam

kageurufu, 2019-04-11:
I'm using a Thinkpad P1 as well, you have to use 400x400 frame size for /dev/video2

Heres my current config

```ini
[core]
detection_notice = false
no_confirmation = false
suppress_unknown = false
ignore_ssh = true
ignore_closed_lid = true
disabled = false
use_cnn = false

[video]
certainty = 3.5
timeout = 4
device_path = /dev/video2
max_height = 400
frame_width = 400
frame_height = 400
dark_threshold = 50
recording_plugin = opencv
device_format = v4l2
force_mjpeg = false
exposure = -1

[debug]
end_report = false
```

clegett, 2019-04-11:
This worked for me. I had to set `dark_threshold` to 100, but otherwise that fixed the problem. Thanks!

closed 2019-04-11 by clegett

kageurufu, 2019-04-11:
Glad to help! What issues were you having with the dark threshold at 50? I'm not sure if I need to adjust the setting myself

boltgolt, 2019-04-11:
Having to up `dark_threshold` usually means all incoming frames are quite dark. This could be caused by the wrong exposure settings or just mean you're further away from the camera. Setting `dark_threshold` to 100 effectively disables dark frame skipping.

clegett, 2019-04-12:
With the default setting of 50 `howdy add` never finished. It just sat there blinking at me. I tried 25 which still didn't work, then 100. Setting it to 100 worked right away.