boltgolt/howdy #148: Login loop in TTY

timgott, 2019-02-14:
I installed Howdy 2.5.0 on Ubuntu 18.10 and it works fine and I can login to the desktop without problems.

If I try to login to a TTY instead of the desktop however, I enter my name and it says my face has been recognized and the Ubuntu welcome text briefly shows up but then the screen returns to the user name prompt.

/var/log/auth.log shows that the session closes again instantly after login:
```
Feb 14 16:29:01 Stahlhand sudo:     root : TTY=tty4 ; PWD=/ ; USER=root ; COMMAND=/usr/bin/timeout 1 loginctl unlock-sessions --no-ask-password
Feb 14 16:29:01 Stahlhand sudo: pam_unix(sudo:session): session opened for user root by LOGIN(uid=0)
Feb 14 16:29:01 Stahlhand sudo: pam_unix(sudo:session): session closed for user root
Feb 14 16:29:01 Stahlhand login[25765]: pam_unix(login:session): session opened for user tim by LOGIN(uid=0)
Feb 14 16:29:01 Stahlhand systemd-logind[915]: New session 26 of user tim.
Feb 14 16:29:01 Stahlhand login[25765]: pam_unix(login:session): session closed for user tim
Feb 14 16:29:01 Stahlhand systemd-logind[915]: Session 26 logged out. Waiting for processes to exit.
Feb 14 16:29:01 Stahlhand systemd-logind[915]: Removed session 26.
```

boltgolt, 2019-02-17:
Please take a look at the fix i just posted in #139, it might fix this issue as well.

timgott, 2019-02-17:
Yes clearing the dismiss_lockscreen option did the trick.
Thanks!

closed 2019-02-17 by boltgolt