boltgolt/howdy #151: Support for Gentoo

lordofserenity, 2019-02-20:
I don't know if this is a lot of work or not, but would it be possible for Howdy to be ported to Gentoo? If that is too much work, could you add documentation so that I could compile from source?

boltgolt, 2019-02-20:
While it should definitely be possible, it probably won't be easy.

There are no compilation steps because Howdy is written in python. A good start would be to check all [dependencies](https://github.com/boltgolt/howdy/blob/master/debian/control#L12) are available and functional on Gentoo. The pam-python package in particular isn't very well supported. The Debian [installation script](https://github.com/boltgolt/howdy/blob/master/debian/postinst) is also quite well documented and should point you in the right direction.

firelizzard18, 2021-05-30:
I created an ebuild for pam-python, and started work on one for howdy. For pam-python, I used gentoo's distutils-r1.class instead of the makefile. I successfully built an egg, but I may be missing a config file or something.

https://github.com/firelizzard18/howdy/tree/gentoo-ebuild

Shifulor, 2021-07-07:
Hi there, you are amazing to tackle that task, may I ask how far you are? Or how I can mybe help? I am moving to gentoo soon and howdy is something i would love to have :)

firelizzard18, 2021-07-07:
I got hung up on figuring out how to write Gentoo ebuilds. The build scripts for Arch Linux are very useful, since Arch and Gentoo are almost cousins, but there's still a fair bit of figuring out to do. Someone experienced with Gentoo ebuilds could probably do this easily - it's not that complicated, especially [howdy/PKGBUILD](https://github.com/boltgolt/howdy/blob/master/archlinux/howdy/PKGBUILD). But I've never written an ebuild outside of this, and it's not something I want to spend time on at the moment.

Shifulor, 2021-07-07:
I fully understand that. If you can put it into a tarbell, i might be able to make it an ebuild. Not experienced but documentation looks good, but ebuilds require .tar

xz-dev, 2023-05-29:
If you need, [there](https://github.com/orsonteodoro/oiledmachine-overlay/tree/master/app-admin/howdy) is already.