boltgolt/howdy #152: USB realsense camera only working in realsense-viewer

boltgolt, 2019-02-20:
I have a related problem. A realsense camera, that is connected via USB.to my systemm (Unbuntu 18.04).

+ it shows up as 2 devices (video0, video1). video0 is IR, video1 works in cheese as RGB, howdy can't use any of them

```
Bus 002 Device 005: ID 8086:0aa5 Intel Corp. 
  iManufacturer           1 Intel(R) RealSense(TM) Camera SR300
  iProduct                2 Intel(R) RealSense(TM) Camera SR300
```

Then I installed the realsense SDK (including special uvcvideo drivers). 
```
$ modinfo uvcvideo | grep "version:" 
version:        1.1.2.realsense-1.3.4
srcversion:     C4FB5E9B212E9C39ED6CD7C
```

Now I can see IR + RGB images/videos in realsense-viewer, but not in cheese, GUVCview or howdy. I am ready to provide additional info if needed.

btw: is there a list of verified working cameras?

_Originally posted by @ju916 in https://github.com/boltgolt/howdy/issues/82#issuecomment-465640006_

boltgolt, 2019-02-20:
I've moved your issue to a new thread because it does not seem to relate to the root user problem in the previous thread. 

What does `sudo howdy test` show as output image?

ju916, 2019-02-21:
The IR control light is actiavted but I get an empty window and a couple of error messages in the terminal:

```
X Error: BadAccess (attempt to access private resource denied) 10
  Extension:    130 (MIT-SHM)
  Minor opcode: 1 (X_ShmAttach)
  Resource id:  0x4400003
X Error: BadShmSeg (invalid shared segment parameter) 128
  Extension:    130 (MIT-SHM)
  Minor opcode: 5 (X_ShmCreatePixmap)
  Resource id:  0x440000d
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x440000e
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x440000e
X Error: BadAccess (attempt to access private resource denied) 10
  Extension:    130 (MIT-SHM)
  Minor opcode: 1 (X_ShmAttach)
  Resource id:  0x15a
X Error: BadShmSeg (invalid shared segment parameter) 128
  Extension:    130 (MIT-SHM)
  Minor opcode: 5 (X_ShmCreatePixmap)
  Resource id:  0x440000f
X Error: BadPixmap (invalid Pixmap parameter) 4
  Major opcode: 54 (X_FreePixmap)
  Resource id:  0x440000e
X Error: BadShmSeg (invalid shared segment parameter) 128
  Extension:    130 (MIT-SHM)
  Minor opcode: 2 (X_ShmDetach)
  Resource id:  0x440000d
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x4400010
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x4400010
select timeout
VIDIOC_DQBUF: Die Ressource ist zur Zeit nicht verfügbar
Traceback (most recent call last):
  File "/usr/local/bin/howdy", line 101, in <module>
    import cli.test
  File "/lib/security/howdy/cli/test.py", line 109, in <module>
    frame = cv2.cvtColor(frame, cv2.COLOR_BGR2GRAY)
cv2.error: OpenCV(4.0.0) /io/opencv/modules/imgproc/src/color.cpp:181: error: (-215:Assertion failed) !_src.empty() in function 'cvtColor'
```
I tried to set the camera resolution and force mjpeg - no change.

johri002, 2019-11-16:
> The IR control light is actiavted but I get an empty window and a couple of error messages in the terminal:
>
> ```
> X Error: BadAccess (attempt to access private resource denied) 10
>   Extension:    130 [...]

Open
`sudo nano /etc/environment`

And add this line:
`QT_X11_NO_MITSHM=1`

boltgolt, 2020-01-07:
Closing because the `cv2.cvtColor` crash is now handled differently and because of inactivity

closed 2020-01-07 by boltgolt