boltgolt/howdy #153: Allow setting camera using the more persistent “/dev/v4l/by-id/” symlink

depadiernos, 2019-02-24:
At boot, my webcams will load in random order. Sometimes the IR webcam is video0, other times it is video2. The symlink in “/dev/v4l/by-id/“ provides a unique and persistent way to always point to the IR webcam regardless of which video* file it is loaded under. 

----

[x] I've searched for similar issues already, and my issue has not been reported yet. 

Linux distribution (if applicable): Arch Linux

Howdy version: 2.5.0

depadiernos, 2019-02-24:
This issue is related to: 
https://askubuntu.com/questions/715333/assign-webcam-to-a-specific-dev-video

boltgolt, 2019-02-24:
Are you sure you're on 2.5.0? Howdy has [used](https://github.com/boltgolt/howdy/blob/master/debian/preinst#L61) `/dev/v4l/by-path/` since 2.4.0.

depadiernos, 2019-02-24:
Ugh, I thought I was on 2.5.0. I am actually on 2.3.1 Closing this issue. Sorry for the noise. :sweat_smile:

closed 2019-02-24 by depadiernos

boltgolt, 2019-02-24:
No problem, arch is the only distro that's still on 2.3.1 because a new PKGBUILD hasn't been made yet.