boltgolt/howdy #154: What type of IR camera should I use

bingws, 2019-02-25:
I intend to purchase an IR camera as Howdy image acquisition device to realize face recognition. But there is little information about Howdy on the Internet. I would like to ask what type of IR camera should I use? What are Howdy's requirements for cameras? Can you recommend some models? Thank you.

Ethiarpus, 2019-02-27:
It should work with anything marketed as "Windows Hello Compatible".

bingws, 2019-02-27:
@Ethiarpus Can howdy do Spoofing Detection?

Ethiarpus, 2019-02-27:
If it's shown a photo printed in IR, it will fail, but a normal photo may not provide enough contrast in the IR spectrum to match the model. Not entirely sure on how Howdy works, but it does use the IR camera if available. [Article on Hello cameras being spoofed with IR photos](https://nakedsecurity.sophos.com/2018/01/02/windows-hello-face-recognition-spoofed-with-photographs/)

bingws, 2019-02-27:
@Ethiarpus By the way, what is the security level of Windows hello? Is there an evaluation report?

boltgolt, 2019-02-27:
With Howdy 2.5.0 all "Windows Hello Compatible" camera's should work just fine, thanks @Ethiarpus 

Howdy does not build a 3D model of you face, like windows does. It is less secure.

closed 2019-02-28 by bingws

ju916, 2019-03-01:
@boltgolt @Ethiarpus  wrote: "With Howdy 2.5.0 all "Windows Hello Compatible should work" 

It's fear not that easy. 
I and apparently others are willing to buy a camera to play with Howdy. 
So I borrowed a cam form a coworker that he is using for "Windows Hello". My Linux system (Ubuntu 18.04 LTS) did not even see the IR device. Then I borrowed a realsense cam from another coworker and only get howdy crashes (see bug report). So that's two attempts, no success. I am lucky, that I did not buy any of those.

So we really need some guidance, what to buy. So I think it would be worth your time to create an expandable and maintained list of "cameras that are known to work with Howdy" -- ideally it would feature the Linux distrubution in use. This list doesn't need to be long. 2-3 good devices that are still available on the market would be a great start.

bye, ju

zeka001, 2019-03-01:
IMHO I think it is really not fare to ask for that, @ju916 .
As far as it seems @boltgolt create this repo for imitating "Windows Hello" on Linux. For me that is more than enough (and I use this opportunity to say thanks @boltgolt ). 
Now it is up to the community to create and maintain that list and if it doesn't work for you camera, try to make it work.

We could already start that list, for example, for me, with a Lenovo T480 (integrated IR camera) and in Linux mint 19 cinnamon, everything works perfect. No need to touch configuration.

Up to @boltgolt to decide if such list will make sense

boltgolt, 2019-03-02:
So far all reported cases of cameras not showing up are because they are using GREY8 encoding, something the Linux kernel simply did not support before 4.19, as noted [here](https://github.com/boltgolt/howdy/issues/88#issuecomment-457708769). This is not an issue with Howdy though, as this project does not aim to write and maintain its own camera drivers. Could you try those cameras with an updated kernel @ju916?

In the meantime I've added a [wiki page](https://github.com/boltgolt/howdy/wiki/Cameras-that-have-been-verified-to-work-with-Howdy) to list working models. Thanks for the kind words @zeka001