boltgolt/howdy #158: No face detected

bingws, 2019-02-28:
_Please describe the issue in as much detail as possible, including any errors and traces_

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable):

Howdy version:

bingws, 2019-02-28:
[image]

bingws, 2019-02-28:
As shown in the figure above, I run "howdy test" and can't see any images. Run "howdy add" to prompt "No face detection, aborting". Below is my config. ini file. I modified it a little, but it works the same as before.
`# Howdy config file
# Press CTRL + X to save in the nano editor

[core]
# Print that face detection is being attempted
detection_notice = false

# Do not print anything when a face verification succeeds
no_confirmation = false

# When a user without a known face model tries to use this script, don't
# show an error but fail silently
suppress_unknown = false

# Disable Howdy in remote shells
ignore_ssh = true

# Auto dismiss lock screen on confirmation
# Will run loginctl unlock-sessions after every auth
# Experimental, can behave incorrectly on some systems
dismiss_lockscreen = false

# Disable howdy in the PAM
# The howdy command will still function
disabled = false

# Use CNN instead of HOG
# CNN model is much more accurate than the HOG based model, but takes much more
# computational power to run, and is meant to be executed on a GPU to attain reasonable speed.
use_cnn = false

[video]
# The certainty of the detected face belonging to the user of the account
# On a scale from 1 to 10, values above 5 are not recommended
certainty = 3.5

# The number of seconds to search before timing out
timeout = 4

# The path of the device to capture frames from
# Should be set automatically by an installer if your distro has one
device_path = /dev/v4l/by-path/pci-0000:00:14.0-usb-0:3.1:1.2-video-index0

# Scale down the video feed to this maximum height
# Speeds up face recognition but can make it less precise
max_height = 320

# Set the camera input profile to this width and height
# The largest profile will be used if set to -1
# Automatically ignored if not a valid profile
frame_width = -1
frame_height = -1

# Because of flashing IR emitters, some frames can be completely unlit
# Skip the frame if the lowest 1/8 of the histogram is above this percentage
# of the total
# The lower this setting is, the more dark frames are ignored
dark_threshold = 100

# The recorder to use. Can be either opencv (default), ffmpeg or pyv4l2.
# Switching from the default opencv to ffmpeg can help with grayscale issues.
recording_plugin = opencv

# Video format used by ffmpeg. Options include vfwcap or v4l2.
# FFMPEG only.
device_format = v4l2

# Force the use of Motion JPEG when decoding frames, fixes issues with YUYV
# raw frame decoding.
# OPENCV only.
force_mjpeg = true

[debug]
# Show a short but detailed diagnostic report in console
end_report = false`

@boltgolt

bingws, 2019-02-28:
I solved the problem, but a new one arose.My IR camera unable to use, "howdy" booted the built-in USB camera.Windows Hello styleThe authentication login was unsuccessful.
@boltgolt

bingws, 2019-02-28:
I looked at the source code to solve the above problem. But I have a new problem. I can log in to "root" user at "Terminal" terminal, but in lock screen mode, I prompt "Unknown error: 1", as shown below.
[image]
How to solve it? Thanks
@boltgolt

bingws, 2019-02-28:
After some exploration, I solved it.

boltgolt, 2019-02-28:
Good to see you've worked it out already. How did you solve it?

bingws, 2019-02-28:
@boltgolt Which mistake do you mean? My method is to look at your previous "issue" and see the source code to solve it.

bingws, 2019-03-01:
@boltgolt I want to consult.Why did I use the built-in USB camera and Intel Realsense S300 respectively and get the same image under the "howdy test" command?

closed 2019-04-23 by bingws

saurabhdhillon, 2019-05-15:
@bingws how did you solved this issue.

bingws, 2019-05-15:
modify the default path of howdy，using your path of  device camera