boltgolt/howdy #160: VIDEOIO ERROR: V4L2: Pixel format of incoming image is unsupported by OpenCV

IosebKoplatadze, 2019-03-01:
VIDEOIO ERROR: V4L2: Pixel format of incoming image is unsupported by OpenCV

Please look straight into the camera
Traceback (most recent call last):
  File "/usr/local/bin/howdy", line 89, in <module>
    import cli.add
  File "/lib/security/howdy/cli/add.py", line 147, in <module>
    gsframe = cv2.cvtColor(frame, cv2.COLOR_BGR2GRAY)
cv2.error: OpenCV(4.0.0) /io/opencv/modules/imgproc/src/color.cpp:181: error: (-215:Assertion failed) !_src.empty() in function 'cvtColor'

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): ubuntu 18.10

Howdy version: latest

AnthonyWharton, 2019-03-01:
Can you post your `config.ini`? If you are unaware that is the contents of the file you can find typing:

`sudo howdy config`

Which I believe you should be able to dump into a terminal with:

`sudo cat /lib/security/howdy/config.ini`

I'm currently looking at some issues that are slightly related, but I can't make any promises!

Gingerglaahk, 2019-03-02:
Having the same issue, ubuntu 18.10, camera - Logitech BRIO 4k.

My config file is - 

`# Howdy config file
# Press CTRL + X to save in the nano editor

[core]
# Print that face detection is being attempted
detection_notice = false

# Do not print anything when a face verification succeeds
no_confirmation = false

# When a user without a known face model tries to use this script, don't
# show an error but fail silently
suppress_unknown = false

# Disable Howdy in remote shells
ignore_ssh = true

# Auto dismiss lock screen on confirmation
# Will run loginctl unlock-sessions after every auth
# Experimental, can behave incorrectly on some systems
dismiss_lockscreen = false

# Disable howdy in the PAM
# The howdy command will still function
disabled = false

# Use CNN instead of HOG
# CNN model is much more accurate than the HOG based model, but takes much more
# computational power to run, and is meant to be executed on a GPU to attain reasonable speed.
use_cnn = false

[video]
# The certainty of the detected face belonging to the user of the account
# On a scale from 1 to 10, values above 5 are not recommended
certainty = 3.5

# The number of seconds to search before timing out
timeout = 4

# The path of the device to capture frames from
# Should be set automatically by an installer if your distro has one
device_path = /dev/v4l/by-path/pci-0000:03:00.0-usb-0:2:1.0-video-index2

# Scale down the video feed to this maximum height
# Speeds up face recognition but can make it less precise
max_height = 320

# Set the camera input profile to this width and height
# The largest profile will be used if set to -1
# Automatically ignored if not a valid profile
frame_width = -1
frame_height = -1

# Because of flashing IR emitters, some frames can be completely unlit
# Skip the frame if the lowest 1/8 of the histogram is above this percentage
# of the total
# The lower this setting is, the more dark frames are ignored
dark_threshold = 50

# The recorder to use. Can be either opencv (default), ffmpeg or pyv4l2.
# Switching from the default opencv to ffmpeg can help with grayscale issues.
recording_plugin = opencv

# Video format used by ffmpeg. Options include vfwcap or v4l2.
# FFMPEG only.
device_format = v4l2

# Force the use of Motion JPEG when decoding frames, fixes issues with YUYV
# raw frame decoding.
# OPENCV only.
force_mjpeg = false

[debug]
# Show a short but detailed diagnostic report in console
end_report = false
`

IosebKoplatadze, 2019-03-02:
I reinstalled it, and chose different camera(not dell xps's infrared camera) and it works now

pixeline, 2019-03-02:
Hello!
same issue here, when launching `sudo howdy add` : 

```
Enter a label for this new model [Initial model] (max 24 characters): test
VIDEOIO ERROR: V4L2: Pixel format of incoming image is unsupported by OpenCV

Please look straight into the camera
Traceback (most recent call last):
  File "/usr/local/bin/howdy", line 89, in <module>
    import cli.add
  File "/lib/security/howdy/cli/add.py", line 147, in <module>
    gsframe = cv2.cvtColor(frame, cv2.COLOR_BGR2GRAY)
cv2.error: OpenCV(4.0.0) /io/opencv/modules/imgproc/src/color.cpp:181: error: (-215:Assertion failed) !_src.empty() in function 'cvtColor'

```
system: elementary OS 5.0
camera: Integrated Camera: Integrated C (usb-0000:00:14.0-6):
	/dev/video0
	/dev/video1
```
The camera's  available pixel formats :
```
v4l2-ctl -d /dev/video0 --list-formats
ioctl: VIDIOC_ENUM_FMT
	Index       : 0
	Type        : Video Capture
	Pixel Format: 'MJPG' (compressed)
	Name        : Motion-JPEG

	Index       : 1
	Type        : Video Capture
	Pixel Format: 'YUYV'
	Name        : YUYV 4:2:2

v4l2-ctl -d /dev/video1 --list-formats
ioctl: VIDIOC_ENUM_FMT
	Index       : 0
	Type        : Video Capture
	Pixel Format: ''
	Name        : 00000032-0002-0010-8000-00aa003
```
Please let me know if you need more information.

AnthonyWharton, 2019-03-02:
@Gingerglaahk Your device path is a bit unusual to me, but I'm not familiar with your system. 
If you test that camera in an app like `cheese` or `fswebcam` do you get greyscale output? I've only ever seen cameras at `/dev/videoX` where `X` is a number.

---------

@pixeline Couple of questions as I'm not the maintainer so I'm just scrapping around for details :stuck_out_tongue: What machine is the integrated camera on, and is this definitely an IR camera? Before tweaking things, can you also try some other back-ends like ffmpeg? You should find that in the config file :smile: 

For example, the output that I get from my IR camera:
```
$ v4l2-ctl -d /dev/video2 --list-formats
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'GREY' (8-bit Greyscale)
```

Failing that we could try change the format that `opencv` uses, that could get a bit more involved though.

pixeline, 2019-03-02:
> @pixeline Couple of questions as I'm not the maintainer so I'm just scrapping around for details  What machine is the integrated camera on, and is this definitely an IR camera?
it's a razer stealth 13 laptop, shipped with windows 10 and windows Hello. So I am positive it's an IR camera ;)

 Before tweaking things, can you also try some other back-ends like `ffmpeg`? You should find that in the config file 
I tried that, and it launched another series of issues :

```
Enter a label for this new model [Initial model] (max 24 characters): test
Traceback (most recent call last):
  File "/usr/local/bin/howdy", line 89, in <module>
    import cli.add
  File "/lib/security/howdy/cli/add.py", line 129, in <module>
    video_capture.grab()
  File "/lib/security/howdy/recorders/ffmpeg_reader.py", line 132, in grab
    self.read()
  File "/lib/security/howdy/recorders/ffmpeg_reader.py", line 105, in read
    self.record(self.numframes)
  File "/lib/security/howdy/recorders/ffmpeg_reader.py", line 89, in record
    .run(capture_stdout=True, quiet=True)
  File "/usr/local/lib/python3.6/dist-packages/ffmpeg/_run.py", line 305, in run
    raise Error('ffmpeg', out, err)
ffmpeg._run.Error: ffmpeg error (see stderr output for detail)
```

AnthonyWharton, 2019-03-02:
Can you see if you get an output when you type `ffmpeg` into a terminal?
Additionally can you run `sudo pip freeze | grep ffmpeg` and paste the output?
It looks like you might not have something you need installed..

boltgolt, 2019-03-02:
You could also try running `dmesg | grep "Unknown video format"`. It's bad news if that command returns anything because it indicates there's simply no driver available to handle the video signal.

pixeline, 2019-03-02:
@boltgolt 
here is the output of `dmesg | grep "Unknown video format"`:
 `[   36.738077] uvcvideo: Unknown video format 00000032-0002-0010-8000-00aa00389b71`

so does it mean that i'm out of luck ?

pixeline, 2019-03-02:
@AnthonyWharton

> Can you see if you get an output when you type `ffmpeg` into a terminal?
> Additionally can you run `sudo pip freeze | grep ffmpeg` and paste the output?
> It looks like you might not have something you need installed..

typing `ffmpeg` in the terminal returns 
```
ffmpeg version 3.4.4-0ubuntu0.18.04.1 Copyright (c) 2000-2018 the FFmpeg developers
...
```
and

```
$ sudo pip freeze | grep ffmpeg
ffmpeg-python==0.1.17
```
Thank you for your help!

boltgolt, 2019-03-02:
`00000032-0002-0010-8000-00aa00389b71` is the new Microsoft 8-bit IR format, supported in Linux kernels starting from 4.17. So updating to a newer kernel will probably fix the problem, but is not always easy.

pixeline, 2019-03-02:
@boltgolt thank you ! 
`uname –sr` indeed  returns `Linux 4.15.0-45-generic`
i'm going to try a kernel update..

AnthonyWharton, 2019-03-02:
@boltgolt The more you know! Thanks :smile: Might be worth updating that in the main README/the message printed when adding a face without a `device_path` set up - if not done already...

------

@pixeline Are you on Ubuntu or Debian? Kernel upgrades aren't too hard, I can point you in the right direction :smile:

pixeline, 2019-03-02:
@AnthonyWharton i'm on elementaryOS, which is based on ubuntu 18.04
i'm currently playing with ukuu, going for 4.19.26... only a reboot away ;)

pixeline, 2019-03-02:
@boltgolt so: kernel upgraded without issue :relieved: 
now, `dmesg | grep "Unknown video format"` returns nothing, and when i try howdy add, i can see the camera is trying to work, but it ends up with this error : 

```
sudo howdy add
Adding face model for the user aplennevaux
Enter a label for this new model [Initial model] (max 24 characters): test
select timeout
VIDIOC_DQBUF: Resource temporarily unavailable

Please look straight into the camera
VIDIOC_QBUF: Invalid argument
Traceback (most recent call last):
  File "/usr/local/bin/howdy", line 89, in <module>
    import cli.add
  File "/lib/security/howdy/cli/add.py", line 147, in <module>
    gsframe = cv2.cvtColor(frame, cv2.COLOR_BGR2GRAY)
cv2.error: OpenCV(4.0.0) /io/opencv/modules/imgproc/src/color.cpp:181: error: (-215:Assertion failed) !_src.empty() in function 'cvtColor'
```

boltgolt, 2019-03-02:
What happens if you open that camera in another application, such as cheese?

pixeline, 2019-03-02:
@boltgolt it works fine. No issue whatsoever.

wzrdtales, 2019-03-03:
Same problem after upgrade.

wzrdtales, 2019-03-03:
A lot on the config change, updating it and setting the device to video0 fixed it.

rohancl, 2019-03-14:
Hey guys, I am a linux newbie and I need help in this, I'm getting the same error. Someone please help me add a face model in howdy

AnthonyWharton, 2019-03-15:
Can you provide some more information?

1. What OS/Distro are you on?
2. Output of `uname -r`
3. Have you worked out your camera device number, and then added it to the howdy config file `device_path` entry? (with `sudo howdy config`)

rohancl, 2019-03-15:
1. ```OS: Ubuntu 16.04``` 
2.  ```4.15.0-46-generic```
3. NO, how do I get the ```device_path``` for now it's ```/dev/v4l/by-path/pci-0000:00:14.0-usb-0:5:1.2-video-index0
```

rohancl, 2019-03-15:
<h1> How is Howdy Failing for me</h1>
@IosebKoplatadze  I tried what you said. "Choosing the other option" It did work, but there is a drawback. The laptop is just using the camera. I can't get Howdy to use both camera and IR sensors at the same time. which basically means Howdy won't be able to recognize you in the dark like windows hello (which uses both IR and camera at the same time for detection). Also Howdy would be less secure not using the IRs because those measure depth and work on the principles of 3D Camera 
<h2> Problem Info</h2>
The laptop isn't using IR sensors (no red flashes), so one isn't actually using the 100% of his/her laptop. 

Windows hello is an integration of IR sensors to measure depth [3D detection] and Camera to gain a picture working collectively for facial recognition and making the best out of your PC. While using Howdy I am either able to use only the IR sensors or only the camera fro facial recognition. not both at the same time

<b>1. Using IR Sensors (You press 'Y' when you see IRs flashing and no camera indicator light ) :</b>
You get this error
```Please look straight into the camera
Traceback (most recent call last):
File "/usr/local/bin/howdy", line 89, in
import cli.add
File "/lib/security/howdy/cli/add.py", line 147, in
gsframe = cv2.cvtColor(frame, cv2.COLOR_BGR2GRAY)
cv2.error: OpenCV(4.0.0) /io/opencv/modules/imgproc/src/color.cpp:181: error: (-215:Assertion failed) !_src.empty() in function 'cvtColor'
```
<b>2. Using Camera (You press 'Y' when you see Camera indicator light but no IRs flashing) :</b>
You don't really get any errors, but then Howdy uses camera only for facial recognition.

--- Please help me, How do I get the IRs and camera to work together ?---

AnthonyWharton, 2019-03-15:
Can you post the output of `v4l2-ctl -d /dev/video0 --list-formats` and `dmesg | grep "Unknown video format"` ?

It looks like you might be having a similar problem to above, where kernels lower than 4.17 do not support the format required. The output of that commands above should give us an idea on that.

---

> Also Howdy would be less secure not using the IRs because those measure depth and work on the principles of 3D Camera 

Not quite, at least to my (limited) knowledge. The IR lights are there to illuminate you, otherwise the frame is extremely dark, even in lighter (visible light) areas. In some sense depth is measured by the brightness, as your face compared to the dark background is very bright as it's right in front of the lights, but this is more just a high contrast/intensity gradient. You should be able to see the output of your camera with `sudo howdy test`, or by using a program like `cheese` - so you can see what it is doing! However, if there's a problem with support for your camera in your current kernel, these may not work.

rohancl, 2019-03-15:
@AnthonyWharton Yeah You're right. I researched a bit and I got to know that the IRs can measure distance, but they don't in windows hello I guess. Thanks for correcting me 
As for the details
[image]
[image]
 Please explain the meanings of the above lines as well. Thanks

AnthonyWharton, 2019-03-15:
Looking at the second output, we see a familiar culprit. As @boltgolt said in a post above:

> `00000032-0002-0010-8000-00aa00389b71` is the new Microsoft 8-bit IR format, supported in Linux kernels starting from 4.17.

So it looks like you need to update your kernel. **If you are unfamiliar you should look up how to do this and the risks behind doing this incorrectly before you start,** I am not responsible for you not being able to boot back into your install :stuck_out_tongue:  

That being said, it's not too difficult, below are the instructions as I recall them (however as I said, look this up before you start as some things may have changed as I haven't used Ubuntu in a while).

The list of Ubuntu Kernels can be found here: https://kernel.ubuntu.com/~kernel-ppa/mainline/. You may as well go for the latest stable version (without an `-rcX` at the end of the name), which at the time of writing is: https://kernel.ubuntu.com/~kernel-ppa/mainline/v5.0.2/.

Then you will want to download 3 files for your architecture, `linux-headers-...generic...`, `linux-image-...generic...` and `linux-modules-...generic...`. 

After that you should install them with `sudo dpkg -i <filename>.deb`. Finally you will need to update grub so it knows about the new kernel version with `sudo update-grub`.

After it's installed the output of `uname -r` should change to the new kernel version, and you should still be able to access the old kernel version under the advanced boot menu items in grub when you boot up.

Warsenius, 2019-04-02:
```
$ v4l2-ctl -d /dev/video0 --list-formats
ioctl: VIDIOC_ENUM_FMT
	Index       : 0
	Type        : Video Capture
	Pixel Format: 'MJPG' (compressed)
	Name        : Motion-JPEG

	Index       : 1
	Type        : Video Capture
	Pixel Format: 'YUYV'
	Name        : YUYV 4:2:2

$ uname -r
4.18.0-17-generic
$ uname -a
Linux 4.18.0-17-generic #18~18.04.1-Ubuntu SMP Fri Mar 15 15:27:12 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
$ dmesg | grep "Unknown video format"
[    1.798572] uvcvideo: Unknown video format 00000032-0002-0010-8000-00aa00389b71
```

I am having the same issue but im already on kernal 4.18

sviatco, 2019-04-06:
After rereading this thread multiple times, it seems that the issue is with IR + Camera available as a separate video input device not functioning, while the camera alone, a dedicated video input, does work.

I have the same repro steps as @rohancl described above and with the XPS 9370 (4.18.0-17-generic) I have 4 video input devices, and if I choose that 3d one during the installation (with the regular webcam light on), howdy does work both for lock screen and general authentication, while using the system, however the IR blasters are never engaged.

I assume that the only downside is that without illumination from IR blasters the face detection will not be usable in the dark. Security-wise it's still the same, since it's comparing a flat bitmap image with OpenCV. Right?

boltgolt, 2019-04-07:
There's also a slight advantage that fooling the face recognition with a printed image is harder in IR, but you're correct in saying that it does not matter that much in terms of security.

AnthonyWharton, 2019-04-08:
@Warsenius You don't have any other devices under `/dev/videoX` do you?

---

@sviatco Intrigued as to why your IR bulbs aren't lighting up as I have the same laptop as you and it's working fine for me. What distro/kernel are you on? Would be interesting to compare and work out why your install doesn't work.

As for what happens without the IR bulbs lighting; you're going to get a lot of inconsistency as to the image that the camera is doing facial recognition against. Differing amounts of IR light from your environment (indoor/outdoor) will affect the results a lot more than if you have a consistent bright source of IR light hitting your face from the direction of your laptop. This light also reflects off of your face a lot more than the surrounding, which gives a more contrasting 'bright' face compared to the darker background when the exposure is taken into account.

sviatco, 2019-04-09:
@AnthonyWharton I'm currently on Ubuntu 18.10 with 4.18.0-17-generic. IR blasters themselves do work, but when I choose that input during howdy's configuration, it produces the error already mentioned above:
```
Traceback (most recent call last):
File "/usr/local/bin/howdy", line 89, in
import cli.add
File "/lib/security/howdy/cli/add.py", line 147, in
gsframe = cv2.cvtColor(frame, cv2.COLOR_BGR2GRAY)
cv2.error: OpenCV(4.0.0) /io/opencv/modules/imgproc/src/color.cpp:181: error: (-215:Assertion failed) !_src.empty() in function 'cvtColor'
```

aroslanov, 2019-04-20:
Same problem for me. 

Laptop Asus Q535UD-BI7T11

$ v4l2-ctl -d /dev/video0 --list-formats
ioctl: VIDIOC_ENUM_FMT
	Index       : 0
	Type        : Video Capture
	Pixel Format: 'MJPG' (compressed)
	Name        : Motion-JPEG

	Index       : 1
	Type        : Video Capture
	Pixel Format: 'YUYV'
	Name        : YUYV 4:2:2

$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.2 LTS
Release:	18.04
Codename:	bionic

$ uname -r
4.18.0-17-generic

$ ffmpeg
ffmpeg version 3.4.4-0ubuntu0.18.04.1 Copyright (c) 2000-2018 the FFmpeg developers

ymatsiuk, 2019-05-12:
> > `00000032-0002-0010-8000-00aa00389b71` is the new Microsoft 8-bit IR format, supported in Linux kernels starting from 4.17. So updating to a newer kernel will probably fix the problem, but is not always easy.
> 
> I had the same problem. Did not work in Ubuntu 18.04 not even with HWE stack and 4.18;
> Installing mainline 4.19 kernel with _ukuu_ fixed that.
> But you are moving away from plain Ubuntu supported kernels with that.

I can confirm this.
Dell XPS 13 (9370) with latest HWE kernel doesn't work properly, face detection is flaky and causes a lot of headache. Especially if the light changes.
Upgrading kernel to mainline latest `5.1.1-050101-generic` fixes the issue with IR blasters and recognition works like a charm.
Steps (do it on your own risk as those are not official packages):
```
wget https://kernel.ubuntu.com/\~kernel-ppa/mainline/v5.1.1/linux-headers-5.1.1-050101_5.1.1-050101.201905110631_all.deb
wget https://kernel.ubuntu.com/\~kernel-ppa/mainline/v5.1.1/linux-headers-5.1.1-050101-generic_5.1.1-050101.201905110631_amd64.deb
wget https://kernel.ubuntu.com/\~kernel-ppa/mainline/v5.1.1/linux-image-unsigned-5.1.1-050101-generic_5.1.1-050101.201905110631_amd64.deb
wget https://kernel.ubuntu.com/\~kernel-ppa/mainline/v5.1.1/linux-modules-5.1.1-050101-generic_5.1.1-050101.201905110631_amd64.deb
sudo dpkg -i *.deb
```

AnthonyWharton, 2019-05-12:
@sviatco Sorry I didn't reply for over a month, been rather busy. 

Have you tried the ffmpeg backend? I've had issues with OpenCV with a pull request I've been working on that get's the same error. This happens when it is failing to read anything from the device specified in the config file, but howdy doesn't report this very well. I've had better luck with ffmpeg in cases where OpenCV fails. 

You'll probably need to install ffmpeg if you don't have it, I can't recall if you need anything else.

krichter722, 2019-05-13:
@AnthonyWharton I'm interested in investigating this issue further. Can you provide information how to change the backend, please. [I tried to do some research](https://github.com/boltgolt/howdy/search?q=backend&type=Issues), but I don't seem to figure it out.

AnthonyWharton, 2019-05-13:
@krichter722 Should be as simple as changing the `recording_plugin` (if I remember the name of the setting correctly) in the config file from `opencv` to `ffmpeg`.

Replying on a phone so it may be slightly different as I'm going from memory.

krichter722, 2019-05-13:
@AnthonyWharton Thanks for you quick reply, I was able to change `recording_plugin` to `ffmpeg` which requested me to install `ffmpeg-python` with `pip3` which I did. `sudo howdy add` now fails due to

```
$ sudo howdy add
Adding face model for the user [obfuscated]
Enter a label for this new model [Initial model] (max 24 characters): test
Traceback (most recent call last):
  File "/usr/local/bin/howdy", line 89, in <module>
    import cli.add
  File "/lib/security/howdy/cli/add.py", line 129, in <module>
    video_capture.grab()
  File "/lib/security/howdy/recorders/ffmpeg_reader.py", line 132, in grab
    self.read()
  File "/lib/security/howdy/recorders/ffmpeg_reader.py", line 105, in read
    self.record(self.numframes)
  File "/lib/security/howdy/recorders/ffmpeg_reader.py", line 79, in record
    self.probe()
  File "/lib/security/howdy/recorders/ffmpeg_reader.py", line 62, in probe
    probe = ffmpeg.probe(self.device_path)
  File "/usr/local/lib/python3.7/dist-packages/ffmpeg/_probe.py", line 19, in probe
    raise Error('ffprobe', out, err)
ffmpeg._run.Error: ffprobe error (see stderr output for detail)
```

I `[obfuscated]` my username. I'm not sure how to proceed. I can file a bug if you want in order to avoid growing this issue beyond its scope.

I can't test the recording plugin `pyv4l2` because `sudo pip3 install pyv4l2` fails due to
<details>
<summary>a large amount of compilation errors</summary>
<pre>
$ sudo pip3 install pyv4l2
WARNING: The directory '/home/richter/.cache/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
WARNING: The directory '/home/richter/.cache/pip' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Collecting pyv4l2
  Downloading https://files.pythonhosted.org/packages/33/de/1f484ce034347d8900923d005e613d49531ec5f9cce48e810c3dff008e8d/pyv4l2-1.0.2.tar.gz (54kB)
     |████████████████████████████████| 61kB 1.2MB/s 
Building wheels for collected packages: pyv4l2
  Building wheel for pyv4l2 (setup.py) ... error
  ERROR: Complete output from command /usr/bin/python3 -u -c 'import setuptools, tokenize;__file__='"'"'/tmp/pip-install-7xpn2syv/pyv4l2/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-io018gij --python-tag cp37:
  ERROR: running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-x86_64-3.7
  creating build/lib.linux-x86_64-3.7/pyv4l2
  copying pyv4l2/exceptions.py -> build/lib.linux-x86_64-3.7/pyv4l2
  copying pyv4l2/__init__.py -> build/lib.linux-x86_64-3.7/pyv4l2
  running build_ext
  building 'pyv4l2/frame' extension
  creating build/temp.linux-x86_64-3.7
  creating build/temp.linux-x86_64-3.7/pyv4l2
  x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.7m -c pyv4l2/frame.c -o build/temp.linux-x86_64-3.7/pyv4l2/frame.o
  pyv4l2/frame.c: In function ‘__Pyx__GetException’:
  pyv4l2/frame.c:3372:24: error: ‘PyThreadState’ {aka ‘struct _ts’} has no member named ‘exc_type’; did you mean ‘curexc_type’?
       tmp_type = tstate->exc_type;
                          ^~~~~~~~
                          curexc_type
  pyv4l2/frame.c:3373:25: error: ‘PyThreadState’ {aka ‘struct _ts’} has no member named ‘exc_value’; did you mean ‘curexc_value’?
       tmp_value = tstate->exc_value;
                           ^~~~~~~~~
                           curexc_value
  pyv4l2/frame.c:3374:22: error: ‘PyThreadState’ {aka ‘struct _ts’} has no member named ‘exc_traceback’; did you mean ‘curexc_traceback’?
       tmp_tb = tstate->exc_traceback;
                        ^~~~~~~~~~~~~
                        curexc_traceback
  pyv4l2/frame.c:3375:13: error: ‘PyThreadState’ {aka ‘struct _ts’} has no member named ‘exc_type’; did you mean ‘curexc_type’?
       tstate->exc_type = local_type;
               ^~~~~~~~
               curexc_type
  pyv4l2/frame.c:3376:13: error: ‘PyThreadState’ {aka ‘struct _ts’} has no member named ‘exc_value’; did you mean ‘curexc_value’?
[... 162 lines cut ...]
                             curexc_value
    pyv4l2/frame.c:3434:22: error: ‘PyThreadState’ {aka ‘struct _ts’} has no member named ‘exc_traceback’; did you mean ‘curexc_traceback’?
         tmp_tb = tstate->exc_traceback;
                          ^~~~~~~~~~~~~
                          curexc_traceback
    pyv4l2/frame.c:3435:13: error: ‘PyThreadState’ {aka ‘struct _ts’} has no member named ‘exc_type’; did you mean ‘curexc_type’?
         tstate->exc_type = type;
                 ^~~~~~~~
                 curexc_type
    pyv4l2/frame.c:3436:13: error: ‘PyThreadState’ {aka ‘struct _ts’} has no member named ‘exc_value’; did you mean ‘curexc_value’?
         tstate->exc_value = value;
                 ^~~~~~~~~
                 curexc_value
    pyv4l2/frame.c:3437:13: error: ‘PyThreadState’ {aka ‘struct _ts’} has no member named ‘exc_traceback’; did you mean ‘curexc_traceback’?
         tstate->exc_traceback = tb;
                 ^~~~~~~~~~~~~
[... cut ...]
ERROR: Command "/usr/bin/python3 -u -c 'import setuptools, tokenize;__file__='"'"'/tmp/pip-install-7xpn2syv/pyv4l2/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-4ckzh831/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-install-7xpn2syv/pyv4l2/
</pre>
</details>
If https://github.com/pupil-labs/pyv4l2 is the official repository the project didn't see any change since 2015. It's build process doesn't make it appear very mature.

I'm running these test on Ubuntu 19.04.

AnthonyWharton, 2019-05-13:
@krichter722 Having had a quick look at the code; the error with `ffmpeg` occurs then `ffprobe` exits with a non-zero status code[1].  Can you please check that you get a response on your command line when you type in `ffmpeg` or `ffprobe` - in other words check that ffmpeg is installed.  

If it is installed you'd need to look deeper into the issue, preferably finding out what `ffprobe` is returning so as to get a few more hints.  Hopefully you can appreciate these aren't my projects so I'm not super familiar with go-to fixes.

As for `pyv4l2`, make sure you have `libv4l` installed: `sudo apt install libv4l-dev`[2]

[1] https://github.com/kkroening/ffmpeg-python/blob/5acc6da7abb428c486a838ac38ed1398c7d524ec/ffmpeg/_probe.py
[2] https://pypi.org/project/pyv4l2/

krichter722, 2019-05-14:
@AnthonyWharton Thanks for your ongoing support.

> As for pyv4l2, it's been updated more recently thatn 2015, it appears that you have a different repo there. You should check PyPi to see what pip is using - and having a cursory look I'd make sure you have libv4l installed: sudo apt install libv4l-dev[2]

Good hint, thank you. The software in the repository https://github.com/duanhongyi/pyv4l2 referenced von PyPi fails to build as well[1].

> Having had a quick look at the code; the error with ffmpeg occurs then ffprobe exits with a non-zero status code[1]. Can you please check that you get a response on your command line when you type in ffmpeg or ffprobe - in other words check that ffmpeg is installed.

`ffmpeg` is installed. What could I use to test my local version for the availability of features needed by `howdy`?
<details>
<summary>The result of ffmpeg and ffprobe command invocation</summary>
<pre>
$ ffmpeg
ffmpeg version 4.1.3-0ubuntu1 Copyright (c) 2000-2019 the FFmpeg developers
  built with gcc 8 (Ubuntu 8.3.0-6ubuntu1)
  configuration: --prefix=/usr --extra-version=0ubuntu1 --toolchain=hardened --libdir=/usr/lib/x86_64-linux-gnu --incdir=/usr/include/x86_64-linux-gnu --arch=amd64 --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared
  libavutil      56. 22.100 / 56. 22.100
  libavcodec     58. 35.100 / 58. 35.100
  libavformat    58. 20.100 / 58. 20.100
  libavdevice    58.  5.100 / 58.  5.100
  libavfilter     7. 40.101 /  7. 40.101
  libavresample   4.  0.  0 /  4.  0.  0
  libswscale      5.  3.100 /  5.  3.100
  libswresample   3.  3.100 /  3.  3.100
  libpostproc    55.  3.100 / 55.  3.100
Hyper fast Audio and Video encoder
usage: ffmpeg [options] [[infile options] -i infile]... {[outfile options] outfile}...

Use -h to get full help or, even better, run 'man ffmpeg'
$ echo $?
1
$ ffprobe
ffprobe version 4.1.3-0ubuntu1 Copyright (c) 2007-2019 the FFmpeg developers
  built with gcc 8 (Ubuntu 8.3.0-6ubuntu1)
  configuration: --prefix=/usr --extra-version=0ubuntu1 --toolchain=hardened --libdir=/usr/lib/x86_64-linux-gnu --incdir=/usr/include/x86_64-linux-gnu --arch=amd64 --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared
  libavutil      56. 22.100 / 56. 22.100
  libavcodec     58. 35.100 / 58. 35.100
  libavformat    58. 20.100 / 58. 20.100
  libavdevice    58.  5.100 / 58.  5.100
  libavfilter     7. 40.101 /  7. 40.101
  libavresample   4.  0.  0 /  4.  0.  0
  libswscale      5.  3.100 /  5.  3.100
  libswresample   3.  3.100 /  3.  3.100
  libpostproc    55.  3.100 / 55.  3.100
Simple multimedia streams analyzer
usage: ffprobe [OPTIONS] [INPUT_FILE]

You have to specify one input file.
Use -h to get full help or, even better, run 'man ffprobe'.
</pre>
</details>

---
[1] I replicated the issue on Travis CI which they can introduce in their project; proposed the script at https://travis-ci.org/krichter722/pyv4l2/builds/532304890 where you can find a link to details about the build failure in case you're interested.

AnthonyWharton, 2019-05-14:
@krichter722 

> Thanks for your ongoing support.

No problem.

> The software in the repository https://github.com/duanhongyi/pyv4l2 referenced von PyPi fails to build as well[1].

Right you are, I thought I had this package already installed on my system but I was incorrect as I never used this backend.  It appears to break for me in the same way when I attempt to install it, so this is likely a problem with the package.  I see you have seen [this issue](https://github.com/duanhongyi/pyv4l2/issues/3) already which suggests it is a GCC version incompatibility.

> ffmpeg is installed. What could I use to test my local version for the availability of features needed by howdy?

Excuse my language but, balls!  Nothing is ever simple in this world.  Having had another dig into [the code](https://github.com/boltgolt/howdy/blob/master/src/recorders/ffmpeg_reader.py#L51) I can see that it is calling the following:

```
ffmpeg -f <device_format> -list_formats all -i <device_path>
```

The two failure cases for this are when the return code is incorrect, or if a regex cannot extract a device resolution (It's looking for two 3 or 4 digit strings of numbers separated by an 'x': `DDD[D]xDDD[D]`)..  After this is falls back to the ffmpeg python library's `probe()` method.  

I suggest you look here to see what it is complaining about.

<details>
<summary>For reference this is what my output looks like (with the settings from my config filled into the above command)</summary>
<br>

```
anthony@demo: $ ffmpeg -f v4l2 -list_formats all -i /dev/video2

ffmpeg version n4.1.3 Copyright (c) 2000-2019 the FFmpeg developers
  built with gcc 8.2.1 (GCC) 20181127
  configuration: --prefix=/usr --disable-debug --disable-static --disable-stripping --enable-fontconfig --enable-gmp --enable-gnutls --enable-gpl --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libdrm --enable-libfreetype --enable-libfribidi --enable-libgsm --enable-libiec61883 --enable-libjack --enable-libmodplug --enable-libmp3lame --enable-libopencore_amrnb --enable-libopencore_amrwb --enable-libopenjpeg --enable-libopus --enable-libpulse --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libv4l2 --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxcb --enable-libxml2 --enable-libxvid --enable-nvdec --enable-nvenc --enable-omx --enable-shared --enable-version3
  libavutil      56. 22.100 / 56. 22.100
  libavcodec     58. 35.100 / 58. 35.100
  libavformat    58. 20.100 / 58. 20.100
  libavdevice    58.  5.100 / 58.  5.100
  libavfilter     7. 40.101 /  7. 40.101
  libswscale      5.  3.100 /  5.  3.100
  libswresample   3.  3.100 /  3.  3.100
  libpostproc    55.  3.100 / 55.  3.100
[video4linux2,v4l2 @ 0x5588964a8e40] Raw       :        gray :      8-bit Greyscale : 340x340
/dev/video2: Immediate exit requested
<exit status 1>
```

As you can see, it pulls out `"340x340"` and returns with exit status 1 as the code is looking for.

</details>

krichter722, 2019-05-17:
After building and installing `pyv4l2` from source with `python3 setup.py build --use-cython && sudo python3 setup.py install --use-cython` and configuring `recording_plugin = pyv4l2` in `/lib/security/howdy/config.ini` I get

```
$ sudo howdy add
Adding face model for the user richter
Enter a label for this new model [Initial model] (max 24 characters): test
Traceback (most recent call last):
  File "/usr/local/bin/howdy", line 89, in <module>
    import cli.add
  File "/lib/security/howdy/cli/add.py", line 109, in <module>
    video_capture = pyv4l2_reader(config.get("video", "device_path"), config.get("video", "device_format"))
  File "/lib/security/howdy/recorders/pyv4l2_reader.py", line 27, in __init__
    self.probe()
  File "/lib/security/howdy/recorders/pyv4l2_reader.py", line 50, in probe
    ret = fcntl.ioctl(vd, v4l2.VIDIOC_G_FMT, fmt)
OSError: [Errno 22] Invalid argument
```

Just in case that's of any help.

sviatco, 2019-05-17:
@AnthonyWharton I haven't tried `ffmpeg` backend nor deviating from the main kernel. Will return back to try again once 19.10 is released.

krichter722, 2019-05-22:
This issue is fixed with version 2.5.1 on a Lenovo ThinkPad T480 with Ubuntu 19.04. I'm uncertain about the version as the issue  happened with 2.5.1 as well. Therefore I suggested #214.

dmafanasyev, 2019-06-07:
Hi all.

I have the same issue with version 2.5.1 of Howdy:

`VIDEOIO ERROR: V4L2: Pixel format of incoming image is unsupported by OpenCV`

If I set path to camera (device_path = /dev/video0 for my case) then howdy works properly for recording_plugin = opencv or recording_plugin = ffmpeg. But if I set path to IR emmiters (device_path = /dev/video2) then for recording_plugin = opencv I face the issue. There is only one way to grab frames from IR emmiters correct: set recording_plugin = ffmpeg.

Asus Zenbook UX433FN
openSUSE Tumbleweed 20190604
Kernel 5.1.5-1-default

wout, 2019-08-12:
I had the same issue on elementaryOS Juno (Ubuntu 18.04) on a Dell XPS 13 (9370), only because I selected the IR camera instead of the video camera during setup. The following question is asked during setup:

```
Did your IR emitters turn on? [y/N]
```

I answered `N`, because the video camera light was on but not the IR emitters. Then the same question is asked again. This time, IR emitters were on, so I answered `Y`. This in fact selects the IR emitters as the camera, which results in the `VIDEOIO ERROR: V4L2: Pixel format of incoming image is unsupported by OpenCV` error.

After reinstalling howdy again, I answered `T` when the question above was asked the first time, and now it's working as it should. It's all a matter of words. :smile:

jackbillstrom, 2019-08-14:
Never got it to work during `18.04.3`, but just upgraded to `19.04`, purge'd it and then `apt install` once again.

First thing I see if that the infrareds are flashing, i enter my choice as `y`.

This is where I get the exact same error as @IosebKoplatadze.

Reinstalled again, change my choice to `n` which starts my webcam LED and enter my second choice as `Y`, and voilá!

terminalninja, 2019-09-05:
> `00000032-0002-0010-8000-00aa00389b71` is the new Microsoft 8-bit IR format, supported in Linux kernels starting from 4.17. So updating to a newer kernel will probably fix the problem, but is not always easy.

Upgrading my kernal on dell xps 9730 - ubuntu 18.04 worked. By default it was using the rgb camera and I had to update the configuration `device_path = /dev/v4l/by-path/pci-0000:00:14.0-usb-0:5:1.2-video-index0` this might change for you not sure. After that it started using the IR camera and the ir blasters also works smooth.

GarvTambi, 2020-07-08:
Sir Unknown error is shown while testing:

[image]

EliuPineda, 2021-02-24:
In order to solve the problem, I have tried It and It has worked for mi on a raspberry pi zero w with raspbian lite

sudo apt-get update
sudo apt-get upgrade
sudo reboot
sudo apt-get install python3-opencv
sudo apt install python3-pip
pip3 install paho-mqtt
pip3 install opencv-python
sudo apt-get install libcblas-dev
sudo apt-get install libhdf5-dev
sudo apt-get install libopencv-dev python-opencv python-picamera
sudo apt-get install libavformat-dev
sudo apt-get install libhdf5-serial-dev
sudo apt-get install libatlas-base-dev

It is a lot of commands and may be redundant but it works.

closed 2024-04-13 by musikid