boltgolt/howdy #162: Installation not working on elementaryos 5.0 (Debian)

pixeline, 2019-03-02:
_Please describe the issue in as much detail as possible, including any errors and traces_
After a (apparently) successful installation, running 'howdy' in the terminal shows this error message

> sudo: howdy: command not found

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): 
elementary OS 5.0
Howdy version: 2.5

pixeline, 2019-03-02:
on ElementaryOS, I needed to do `sudo apt-get install software-properties-common` before `sudo add-apt-repository ppa:boltgolt/howdy`

closed 2019-03-02 by pixeline

boltgolt, 2019-03-02:
Thanks for figuring it out!

@rileyball would installing `software-properties-common`also enable PPAs for vanilla debian?

belosthomas, 2019-07-24:
Same issue on Pop!_OS 19.04. Command not found.
`sudo apt-get install software-properties-common` solved nothing for me.

rileyball, 2019-08-02:
@belosthomas @boltgolt You shouldn't enable PPAs on Debian. Doing so is typically called making a "franken-debian" and does more harm than good to your system.