boltgolt/howdy #164: in compare.py

Sanjay126, 2019-03-07:
_Please describe the issue in as much detail as possible, including any errors and traces_

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable):

Howdy version:

boltgolt, 2019-03-07:
What does the following command return?

```
sudo /usr/bin/env python3 -c "import cv2; print(cv2.__version__);"
```

Sanjay126, 2019-03-07:
4.0.1

boltgolt, 2019-03-07:
Looks like you have an functioning openCV install and python3 bindings, but they're called incorrectly. Could you try replacing `"/usr/bin/python3"` with `"/usr/bin/env", "python3"` on line 33 in `/lib/security/howdy/pam.py`?

Sanjay126, 2019-03-07:
Actually I solved it. I ran pip install opencv-python inside compare.py(because error was in that file) once and then removed it.
Thanks for help

closed 2019-03-07 by Sanjay126

CybernetiX-S3C, 2019-04-01:
I am having this issue.

sudo /usr/bin/env python3 -c "import cv2; print(cv2.__version__);"

Traceback (most recent call last):
  File "<string>", line 1, in <module>
ImportError: No module named 'cv2'

Yet with python version 2.7

sudo /usr/bin/env python -c "import cv2; print(cv2.__version__);"

4.0.0

Kali Linux 2019.1
Debian Distro
Python Versions:
[ CybernetiX-S3C howdy ]# python -V
Python 2.7.16rc1
[ CybernetiX-S3C howdy ]# python3 -V
Python 3.5.6

boltgolt, 2019-04-01:
Did you run `pip3 install opencv-python`?

CybernetiX-S3C, 2019-04-02:
Of course I did. I also went to the website to install 4.0 manually from
sources.

NescobarAlopLop, 2019-12-18:
for me installing opencv worked, but not with pip and specifically for python3:
`sudo apt-get install python3-opencv`

Zocker1999NET, 2020-04-02:
Then the issue introduced by @CybernetiX-S3C, and solved by @NescobarAlopLop should be fixed by adding `python3-opencv` to the dependencies of the debian package. Because the package is available in the repositories of Debian (since buster) and Ubuntu (since 19.10 eoan) this should be no problem for current users and make the installation easier for new users.

CybernetiX-S3C, 2020-04-04:
It is available. That is correct. But my error remains. I'd like to also
get rid of the whole not being able to use root.

John Poli Modica
Cyber Security Professional
CybernetiX S3C
https://CybernetiX-S3C.net/

CybernetiX-S3C, 2020-04-04:
Also, as a reminder, I am using the latest Kali Linux with everything up to
date.

John Poli Modica
Cyber Security Professional
CybernetiX S3C
https://CybernetiX-S3C.net/