boltgolt/howdy #165: Howdy segmentation fault QObject::moveToThread

mhstnsc, 2019-03-08:
* install howdy with ppa
* sudo howdy add
* sudo howdy test
```
Opening a window with a test feed                                                                                                                                                                           
                                                                                                                                                                                                            
Press ctrl+C in this terminal to quit                                                                                                                                                                       
Click on the image to enable or disable slow mode                                                                                                                                                           
                                                                                                                                                                                                            
QObject::moveToThread: Current thread (0x3053e20) is not the object's thread (0x3105a30).                                                                                                                   
Cannot move to target thread (0x3053e20)                                                                                                                                                                    
                                                                                                                                                                                                            
Segmentation fault            
```

here is the stack trace
```
#0  0x00007fbd8d7cc1ce in xcb_send_request_with_fds64 () from /usr/lib/x86_64-linux-gnu/libxcb.so.1
[Current thread is 1 (Thread 0x7fbd985e3740 (LWP 21508))]
(gdb) bt
#0  0x00007fbd8d7cc1ce in xcb_send_request_with_fds64 () at /usr/lib/x86_64-linux-gnu/libxcb.so.1
#1  0x00007fbd8d7cc729 in xcb_send_request () at /usr/lib/x86_64-linux-gnu/libxcb.so.1
#2  0x00007fbd8d7d2be4 in xcb_intern_atom () at /usr/lib/x86_64-linux-gnu/libxcb.so.1
#3  0x00007fbd736b405a in  () at /usr/lib/kde4/plugins/styles/breeze.so
#4  0x00007fbd736b415a in  () at /usr/lib/kde4/plugins/styles/breeze.so
#5  0x00007fbd736b4302 in  () at /usr/lib/kde4/plugins/styles/breeze.so
#6  0x00007fbd736c628e in  () at /usr/lib/kde4/plugins/styles/breeze.so
#7  0x00007fbd736d3ec5 in  () at /usr/lib/kde4/plugins/styles/breeze.so
#8  0x00007fbd90c4f4c1 in QStyleFactory::create(QString const&) () at /usr/local/lib/python3.6/dist-packages/cv2/.libs/libQtGui-6d0f14dd.so.4.8.7
#9  0x00007fbd908fd318 in QApplication::style() () at /usr/local/lib/python3.6/dist-packages/cv2/.libs/libQtGui-6d0f14dd.so.4.8.7
#10 0x00007fbd908fd4e5 in QApplicationPrivate::initialize() () at /usr/local/lib/python3.6/dist-packages/cv2/.libs/libQtGui-6d0f14dd.so.4.8.7
#11 0x00007fbd908fd5e0 in QApplicationPrivate::construct(_XDisplay*, unsigned long, unsigned long) () at /usr/local/lib/python3.6/dist-packages/cv2/.libs/libQtGui-6d0f14dd.so.4.8.7
#12 0x00007fbd908fd84f in QApplication::QApplication(int&, char**, int) () at /usr/local/lib/python3.6/dist-packages/cv2/.libs/libQtGui-6d0f14dd.so.4.8.7
#13 0x00007fbd943fccc8 in  () at /usr/local/lib/python3.6/dist-packages/cv2/cv2.cpython-36m-x86_64-linux-gnu.so
#14 0x00007fbd94407335 in  () at /usr/local/lib/python3.6/dist-packages/cv2/cv2.cpython-36m-x86_64-linux-gnu.so
#15 0x00007fbd943f36ea in  () at /usr/local/lib/python3.6/dist-packages/cv2/cv2.cpython-36m-x86_64-linux-gnu.so
#16 0x00007fbd938950b3 in  () at /usr/local/lib/python3.6/dist-packages/cv2/cv2.cpython-36m-x86_64-linux-gnu.so
#17 0x00000000005030d5 in  ()
#18 0x0000000000506859 in _PyEval_EvalFrameDefault ()
#19 0x0000000000504c28 in  ()
#20 0x0000000000511d78 in  ()
#21 0x000000000056617e in PyCFunction_Call ()
#22 0x000000000050bb66 in _PyEval_EvalFrameDefault ()
#23 0x0000000000504c28 in  ()
#24 0x0000000000502540 in  ()
#25 0x0000000000502f3d in  ()
#26 0x0000000000506859 in _PyEval_EvalFrameDefault ()
#27 0x0000000000502209 in  ()
#28 0x0000000000502f3d in  ()
#29 0x0000000000506859 in _PyEval_EvalFrameDefault ()
#30 0x0000000000502209 in  ()
#31 0x0000000000502f3d in  ()
#32 0x0000000000506859 in _PyEval_EvalFrameDefault ()
#33 0x0000000000502209 in  ()
#34 0x0000000000502f3d in  ()
#35 0x0000000000506859 in _PyEval_EvalFrameDefault ()
#36 0x0000000000501945 in _PyFunction_FastCallDict ()
#37 0x00000000005a36f1 in _PyObject_FastCallDict ()
#38 0x000000000059662e in _PyObject_CallMethodIdObjArgs ()
#39 0x00000000004ee84d in PyImport_ImportModuleLevelObject ()
#40 0x000000000050896c in _PyEval_EvalFrameDefault ()
#41 0x0000000000504c28 in  ()
#42 0x0000000000506393 in PyEval_EvalCode ()
#43 0x0000000000634d52 in  ()
#44 0x0000000000634e0a in PyRun_FileExFlags ()
#45 0x00000000006385c8 in PyRun_SimpleFileExFlags ()
#46 0x000000000063915a in Py_Main ()
#47 0x00000000004a6f10 in main ()

```                                                                                                                                                                              

Terminal dump attached

----

Linux distribution (if applicable):
ubuntu-18.04

Howdy version:
Unpacking howdy (2.5.0)

boltgolt, 2019-03-08:
Thanks for reporting, but please search for similar issues first. Looks like a duplicate of #54.

closed 2019-03-08 by boltgolt

mhstnsc, 2019-03-08:
> Thanks for reporting, but please search for similar issues first. Looks like a duplicate of #54.

I did but the stupid search did not give me anything