boltgolt/howdy #168: Dosen't work on Mint Cinamon 19 dlib issue

KaitenIV, 2019-03-12:
I have installed all correctly, but when i try to run "sudo howdy add", terminal give to me an error:

`zenek@zenek:~$ sudo howdy add
Traceback (most recent call last):
  File "/lib/security/howdy/compare.py", line 18, in <module>
    import dlib
ModuleNotFoundError: No module named 'dlib'
Unknown error: 1
[sudo] password for zenek:         
No module named 'dlib'

Can't import the dlib module, check the output of
pip3 show dlib`

I already have tryed to install the dlib and dosent work ... can someone help me ?

boltgolt, 2019-03-12:
What's the output of `pip3 show dlib`?

KaitenIV, 2019-03-13:
Nothing, just return zenek$zenek.

Thanks for reply!

boltgolt, 2019-03-13:
Could run `pip3 install dlib` and try `sudo howdy add` after that?

KaitenIV, 2019-03-14:
zenek@zenek:~$ pip3 install dlib
Collecting dlib
  Using cached
https://files.pythonhosted.org/packages/05/57/e8a8caa3c89a27f80bc78da39c423e2553f482a3705adc619176a3a24b36/dlib-19.17.0.tar.gz
Installing collected packages: dlib
  Running setup.py install for dlib ... error
    Complete output from command /usr/bin/python3 -u -c "import setuptools,
tokenize;__file__='/tmp/pip-install-6gh5_61l/dlib/setup.py';f=getattr(tokenize,
'open', open)(__file__);code=f.read().replace('\r\n',
'\n');f.close();exec(compile(code, __file__, 'exec'))" install --record
/tmp/pip-record-tjggz1so/install-record.txt
--single-version-externally-managed --compile:
    running install
    running build
    running build_py
    package init file 'dlib/__init__.py' not found (or not a regular file)
    running build_ext
    Building extension for Python 3.6.7 (default, Oct 22 2018, 11:32:17)
    Invoking CMake setup: 'cmake
/tmp/pip-install-6gh5_61l/dlib/tools/python
-DCMAKE_LIBRARY_OUTPUT_DIRECTORY=/tmp/pip-install-6gh5_61l/dlib/build/lib.linux-x86_64-3.6
-DPYTHON_EXECUTABLE=/usr/bin/python3 -DCMAKE_BUILD_TYPE=Release'
    -- The C compiler identification is GNU 7.3.0
    -- The CXX compiler identification is GNU 7.3.0
    -- Check for working C compiler: /usr/bin/cc
    -- Check for working C compiler: /usr/bin/cc -- works
    -- Detecting C compiler ABI info
    -- Detecting C compiler ABI info - done
    -- Detecting C compile features
    -- Detecting C compile features - done
    -- Check for working CXX compiler: /usr/bin/c++
    -- Check for working CXX compiler: /usr/bin/c++ -- works
    -- Detecting CXX compiler ABI info
    -- Detecting CXX compiler ABI info - done
    -- Detecting CXX compile features
    -- Detecting CXX compile features - done
    -- Found PythonInterp: /usr/bin/python3 (found version "3.6.7")
    -- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.6m.so
    -- Performing Test HAS_CPP14_FLAG
    -- Performing Test HAS_CPP14_FLAG - Success
    -- pybind11 v2.2.2
    -- Using CMake version: 3.10.2
    -- Compiling dlib version: 19.17.0
    -- SSE4 instructions can be executed by the host processor.
    -- AVX instructions can be executed by the host processor.
    -- Enabling AVX instructions
    -- Looking for pthread.h
    -- Looking for pthread.h - found
    -- Looking for pthread_create
    -- Looking for pthread_create - not found
    -- Looking for pthread_create in pthreads
    -- Looking for pthread_create in pthreads - not found
    -- Looking for pthread_create in pthread
    -- Looking for pthread_create in pthread - found
    -- Found Threads: TRUE
    -- Looking for XOpenDisplay in
/usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so
    -- Looking for XOpenDisplay in
/usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so -
found
    -- Looking for gethostbyname
    -- Looking for gethostbyname - found
    -- Looking for connect
    -- Looking for connect - found
    -- Looking for remove
    -- Looking for remove - found
    -- Looking for shmat
    -- Looking for shmat - found
    -- Looking for IceConnectionNumber in ICE
    -- Looking for IceConnectionNumber in ICE - found
[... 193 lines cut ...]
CMakeFiles/dlib_python.dir/src/global_optimization.cpp.o
    [ 95%] Building CXX object
CMakeFiles/dlib_python.dir/src/image_dataset_metadata.cpp.o
    [ 96%] Building CXX object
CMakeFiles/dlib_python.dir/src/numpy_returns.cpp.o
    [ 97%] Building CXX object CMakeFiles/dlib_python.dir/src/line.cpp.o
    [ 98%] Building CXX object CMakeFiles/dlib_python.dir/src/gui.cpp.o
    [100%] Linking CXX shared module
/tmp/pip-install-6gh5_61l/dlib/build/lib.linux-x86_64-3.6/
dlib.cpython-36m-x86_64-linux-gnu.so
    [100%] Built target dlib_python
    running install_lib
    copying build/lib.linux-x86_64-3.6/dlib.cpython-36m-x86_64-linux-gnu.so
-> /usr/local/lib/python3.6/dist-packages
    error: could not create '/usr/local/lib/python3.6/dist-packages/
dlib.cpython-36m-x86_64-linux-gnu.so': Permission denied

    ----------------------------------------
Command "/usr/bin/python3 -u -c "import setuptools,
tokenize;__file__='/tmp/pip-install-6gh5_61l/dlib/setup.py';f=getattr(tokenize,
'open', open)(__file__);code=f.read().replace('\r\n',
'\n');f.close();exec(compile(code, __file__, 'exec'))" install --record
/tmp/pip-record-tjggz1so/install-record.txt
--single-version-externally-managed --compile" failed with error code 1 in
/tmp/pip-install-6gh5_61l/dlib/

boltgolt, 2019-03-14:
Don't forget the sudo at the start: `sudo pip3 install dlib`

KaitenIV, 2019-03-18:
Finally fix it !

closed 2019-03-26 by boltgolt

faliure, 2019-08-29:
I was missing g++ to install dlib. So...

sudo apt-get install build-essential
sudo pip3 install dlib
sudo howdy add -U my_user

karanjot21, 2019-08-29:
VIDEOIO ERROR: V4L2: Pixel format of incoming image is unsupported by OpenCV

Please look straight into the camera
Traceback (most recent call last):
  File "/usr/local/bin/howdy", line 89, in <module>
    import cli.add
  File "/lib/security/howdy/cli/add.py", line 147, in <module>
    gsframe = cv2.cvtColor(frame, cv2.COLOR_BGR2GRAY)
cv2.error: OpenCV(4.1.0) /io/opencv/modules/imgproc/src/color.cpp:182: error: (-215:Assertion failed) !_src.empty() in function 'cvtColor'

Error in sys.excepthook:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/apport_python_hook.py", line 145, in apport_excepthook
    os.O_WRONLY | os.O_CREAT | os.O_EXCL, 0o640), 'wb') as f:
FileNotFoundError: [Errno 2] No such file or directory: '/var/crash/_lib_security_howdy_cli.py.0.crash'

Original exception was:
Traceback (most recent call last):
  File "/usr/local/bin/howdy", line 89, in <module>
    import cli.add
  File "/lib/security/howdy/cli/add.py", line 147, in <module>
    gsframe = cv2.cvtColor(frame, cv2.COLOR_BGR2GRAY)
cv2.error: OpenCV(4.1.0) /io/opencv/modules/imgproc/src/color.cpp:182: error: (-215:Assertion failed) !_src.empty() in function 'cvtColor'

kiyokodyele, 2019-09-28:
> I was missing g++ to install dlib. So...
> 
> sudo apt-get install build-essential
> sudo pip3 install dlib
> sudo howdy add -U my_user

Worked for me running Linux Mint 19 w/ Logitech Brio

pwozniak89, 2020-03-13:
Thank @faliure that helps me !

agridev, 2020-03-30:
Helllo 
Thank you for this app
I succeeded to install after following all the recommendation as I got the same errors.
I am newbie in Linux envir :)
I run the add for face and  the cam is working and adding face.
The problem that when logging the cam is on and seems flashing my face but there is no longin, I need to add my password to log to my session.
Please advise 
thanks

pwozniak89, 2020-03-30:
Hi, @agridev 
Try to add multiple capture of your face with **howdy add**
Be careful with backlight when camera is recognizing you (lot's of failure for me when a window is behind me)
Does it works when you run sudo on command line ?

agridev, 2020-03-31:
> Hi, @agridev
> Try to add multiple capture of your face with **howdy add**
> Be careful with backlight when camera is recognizing you (lot's of failure for me when a window is behind me)
> Does it works when you run sudo on command line ?

Hi @pwozniak89 
Thank you for the reply.
Yes I added a lot of faces by the sudo command to add.
I do not understand what do you mean by "Does it works when you run sudo on command line ?" does this mean when adding faces? 
Yes it works to add faces, it works to detect faces when authorization to add software or any other authentification request needed, but only thing not working is logging to sessions.
bests 

P.S I am not so expert in command line and linux I am back to linux since year and would like to forget windows (the only thing that makes me installing windows is formatting issues for word documents when exchanged with thirdparties)

Gerlo5, 2020-06-22:
> > I was missing g++ to install dlib. So...
> > sudo apt-get install build-essential
> > sudo pip3 install dlib
> > sudo howdy add -U my_user
> 
> Worked for me running Linux Mint 19 w/ Logitech Brio

Worked for me too (Linux Mint 19.3 w/ HP Zbook, dlib-19.20.0).
(btw) great work!