boltgolt/howdy #169: spurious "sudo" prompts after unlock

ju916, 2019-03-12:
I tested howdy with xscreenlock and after sucessfull unlocks I saw a "sudo" prompt on my console, that had no right to be there. I tracked it down to the following code in pam.py:

```
       # Try to dismiss the lock screen if enabled
       if config.get("core", "dismiss_lockscreen"):
               # Run it as root with a timeout of 1s, and never ask for a password through the UI
               subprocess.Popen(["sudo", "timeout", "1", "loginctl", "unlock-sessions", "--no-ask-password"])
```
My config file says (via default):
`dismiss_lockscreen = false
`
From looking at the rest of the code (I never learned Python!) I figured, the if-statement should check `config.getboolean`. So I changed:

```
54c54
< 		if config.get("core", "dismiss_lockscreen"):
---
> 		if config.getboolean("core", "dismiss_lockscreen"):

```

That fixed the issue for me. Take it with a grain of salt...

boltgolt, 2019-03-12:
This has already been reported in #139, with the known fix also listed. It's a very simple bug with no bad intentions behind it, as you could have easily noticed yourself by looking at the actual native command it was running.

closed 2019-03-12 by boltgolt

ju916, 2019-03-13:
Thanks for the quick answer!

Sorry that I did not see that old issue.
I checked the current code and found that the problem persists in the current release version. 
As the fix is simple and the issue affects every installation (the sudo command is called every single time, face recognition suceeds!) I implied this would have been fixed, if it was known.

btw: I think calling sudo inside of an authentication module is a bad idea - but I will open another issue for that.