boltgolt/howdy #170: calling sudo in a pam module is a bad idea

ju916, 2019-03-13:
I found that pam.py calls sudo after face recognition suceeds. I think that this is a bad idea and should be removed.

Is it really just me who thinks that this is a bad? 
pam.py is a general purpose authentication module, that is called in common-auth -- i.e. on every single authentication request. sudo itself requires authentication (and of course includes common-auth in its pam configuration). So we quickly end up in sudo calling sudo calling sudo ...

More general: calling something that requires authentication inside of an authentication module will lead to unspecified and unpredictable behaviour.

bye, ju

PS: this is not directly related to [issue 139](https://github.com/boltgolt/howdy/issues/139) that can be easily fixed.

boltgolt, 2019-03-13:
I fully agree with you that an authentication module should not try to call `sudo` _by default_. However, the purpose of this program is to mimic Windows Hello on Linux. Using Hello from the lock screen on Windows will automatically dismiss it, so it makes sense for Howdy to implement that too.

At the time of writing that code, the only way to do so was to call `loginctl` with sudo, as it did not work without it. This does not mean it will start a notification loop normally though. Calling sudo within an already authenticated session will not result in a new auth request. You can try this for yourself very easily by running `sudo sudo true`, which should only ask for your password once.

> More general: calling something that requires authentication inside of an authentication module will lead to **unspecified and unpredictable behaviour**.

It's stated [quite clearly](https://github.com/boltgolt/howdy/blob/master/src/config.ini#L20) in the config that this is an experimental option, and can behave incorrectly on some systems. The config option should be set to false on new installations.

ju916, 2019-03-13:
No, what I am saying is: an authetication module should *never* call sudo. It will *always* be a hazard.
I see the problem you are trying to solve. But 

1) in pam.py you know nothing about the authentication state. In general your session is not authenticated for using sudo. It might be - but it also might not. You don't even know if the current authentication process will suceed or not -- it depends on your PAM setup (think 2FA). That's what I am meaning with undefined behaviour. You will get your endless auth loop eventually.

2) what you want to do is: execute a privileged command from a potentially unprivileged process without doing authentication. That is what Set-UID is for. So the proper way to do that would be to create a small setuid-root app, that runs your loginctl-command as root and can be called safely from pam.py,

A word of wanrning though: creating safe setuid apps, that can not be abused for privilege escalation is not an easy task. I would vouch against installing one in a default installation if most people will not need it.

boltgolt, 2019-03-13:
I don't think problems with it are as widespread as you think. With the accidental enabling of this option only a handful of people have viewed the issue thread, while more than 3000 installations have been installed since 2.5.0 rolled out. So far almost every installation I've seen has not re-auth loop. Does make a little bit of sense, seeing as PAM modules run with upped permissions.

In my opinion it's worth the ease of use for people that want to enable it.

ju916, 2019-03-13:
I can't argue with numbers because I don't have any. I can only argue with logic:

1) where exactly should the authentication for running sudo without a new authentication come from? After the system has not beend used for an hour (like in a screenlock state) all sudo auth tokens will have expired.

2) Having an authentication module that has undefined behaviour by design in some real world situations really is acceptable for you?

3) What are the arguments against doing it proper - using setuid?

boltgolt, 2019-03-13:
1. PAM modules are running as semi-root (can read files only root can open), which is enough for sudo to succeed. It does so on my system and most others so far. You'd think that would be enough for loginctl to work without sudo, but that's not the case.

2. It's not "undefined behavior", it either works or loops 5 auths before being killed. Because it's run as a subprocess it has no access to the PAM thread. It's clearly marked as experiential, _normally_ off by default and unexploitable (unless you can think of one). The only real problem i could see here is an UX issue.

3. You've already said the most important one yourself: 

> creating safe setuid apps, that can not be abused for privilege escalation is not an easy task

boltgolt, 2019-03-29:
Fixed in 2.5.1, as auto dismiss of the lockscreen has been removed

closed 2019-03-29 by boltgolt