boltgolt/howdy #171: VIDIOC_QBUF: Invalid argument

pixeline, 2019-03-15:
_Please describe the issue in as much detail as possible, including any errors and traces_
when trying to add a first face recognition (howdy add) i get this error:

``` 
sudo howdy add
Adding face model for the user aplennevaux
Enter a label for this new model [Initial model] (max 24 characters): test
select timeout
VIDIOC_DQBUF: Resource temporarily unavailable

Please look straight into the camera
VIDIOC_QBUF: Invalid argument
Traceback (most recent call last):
  File "/usr/local/bin/howdy", line 89, in <module>
    import cli.add
  File "/lib/security/howdy/cli/add.py", line 147, in <module>
    gsframe = cv2.cvtColor(frame, cv2.COLOR_BGR2GRAY)
cv2.error: OpenCV(4.0.0) /io/opencv/modules/imgproc/src/color.cpp:181: error: (-215:Assertion failed) !_src.empty() in function 'cvtColor'
```
the kernel :
```
uname -r
4.20.13-042013-generic
```
webcam works fine with software like Cheese.

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): elementary OS 5.0

Howdy version: 2.5.0

boltgolt, 2019-03-26:
Could you run `v4l2-ctl -d /dev/video0 --list-formats`, and replace video0 with your IR camera?

adigupta13, 2019-04-12:
Hi, having the same issue,
output for ```v4l2-ctl -d /dev/video0 --list-formats```

ioctl: VIDIOC_ENUM_FMT
	Index       : 0
	Type        : Video Capture
	Pixel Format: 'MJPG' (compressed)
	Name        : Motion-JPEG

	Index       : 1
	Type        : Video Capture
	Pixel Format: 'YUYV'
	Name        : YUYV 4:2:2

adigupta13, 2019-04-12:
I am running linux 4.19 which supports the new microsoft 8bit IR emitter, on the Razer Blade 15 which comes with windows hello

adigupta13, 2019-04-12:
```$dmesg | grep "Unknown video format"``` output is nothing, while install the IR light flashes. So I think it the camera is detected

r-or, 2019-05-05:
I had the same problem. A quick test showed that opencv by default uses a size of 640x360 for this camera. This doesn't work though, it has to be set to 360x360.

So to resolve this issue, you need to manually setup howdy for this resolution.

This can be done via `sudo howdy config`. There you will find `frame_width` and `frame_height`. Set both to 360, then it will work.

pixeline, 2019-05-05:
> I had the same problem. A quick test showed that opencv by default uses a size of 640x360 for this camera. This doesn't work though, it has to be set to 360x360.
>
> So to resolve this issue, you need [...]
awesome, it now works, thank you!

closed 2019-05-05 by pixeline

Young-devps, 2020-05-15:
i have same on raspberry pi when i lauched the google coco stater object detection model

maham-khan999, 2021-04-29:
> i have same on raspberry pi when i lauched the google coco stater object detection model

did u find the solution?? i have the same problem on raspberry pi for detection model. if you find the solution plz help me