boltgolt/howdy #172: Unable to install on Linux Mint 18.3 Sylvia

paveltrufi, 2019-03-18:
I tried to run the commands described in the README (add the repo, update and install), but this is the error I get:
```bash
Preparando para desempaquetar .../archives/howdy_2.5.0_all.deb ...
Starting IR camera check...

Trying "UVC Camera (046d:0825)"
One of your cameras should now be on.
Did your IR emitters turn on? [y/N]: y

Desempaquetando howdy (2.5.0) ...
Procesando disparadores para man-db (2.7.5-1) ...
Configurando howdy (2.5.0) ...

>>> Upgrading pip to the latest version

Traceback (most recent call last):
  File "/usr/bin/pip3", line 9, in <module>
    from pip import main
ImportError: cannot import name 'main'
Error while running last command
dpkg: error al procesar el paquete howdy (--configure):
 el subproceso instalado el script post-installation devolvió el código de salida de error 1
Se encontraron errores al procesar:
 howdy
E: Sub-process /usr/bin/dpkg returned an error code (1)
```
Linux distribution (if applicable): Linux Mint 18.3 Sylvia (4.15.0-45-generic)
Howdy version: the last one available in the PPA?
pip3 version: pip 18.0 from /home/i2e23/.local/lib/python3.5/site-packages/pip (python 3.5)

boltgolt, 2019-03-18:
That's an internal pip error, could you try upgrading pip? `pip3 install --upgrade pip`

paveltrufi, 2019-03-18:
I upgraded pip, but it didn't work:
```bash
[...]
Configurando howdy (2.5.0) ...

>>> Upgrading pip to the latest version

Traceback (most recent call last):
  File "/usr/bin/pip3", line 9, in <module>
    from pip import main
ImportError: cannot import name 'main'
Error while running last command
dpkg: error al procesar el paquete howdy (--configure):
 el subproceso instalado el script post-installation devolvió el código de salida de error 1
Se encontraron errores al procesar:
 howdy
E: Sub-process /usr/bin/dpkg returned an error code (1)
$ pip3 --version                      8.6s  lun 18 mar 2019 15:29:44 CET
pip 19.0.3 from /home/i2e23/.local/lib/python3.5/site-packages/pip (python 3.5)
```

boltgolt, 2019-03-26:
It's an environment issue that collides with the renaming of `pip.main` to `pip._internal.main`, could you try some solutions from [here](https://github.com/pypa/pip/issues/5240)?

paveltrufi, 2019-03-26:
https://github.com/pypa/pip/issues/5240#issuecomment-381677898 worked for me. Thanks!

closed 2019-03-26 by boltgolt