boltgolt/howdy #174: [FEDORA 29] Problem with howdy add

BedoM, 2019-03-21:
_Please describe the issue in as much detail as possible, including any errors and traces_

Hey ! 

[mabed@legiony520 /]$ sudo howdy add
[sudo] Mot de passe de mabed : 
Data files have not been downloaded, please run the following commands:

	cd /usr/lib64/security/howdy/dlib-data
	sudo ./install.sh

[mabed@legiony520 /]$ cd /usr/lib64/security/howdy/dlib-data
bash: cd: /usr/lib64/security/howdy/dlib-data: No such file or directory
[mabed@legiony520 /]$ sudo ./install.sh
sudo: ./install.sh : commande introuvable

What is the problem ?

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): fedora latest stable

Data files have not been downloaded, please run the following commands:

Linux legiony520 4.20.16-200.fc29.x86_64 #1 SMP Thu Mar 14 15:10:22 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

Howdy version:

miba358, 2019-03-26:
> _Please describe the issue in as much detail as possible, including any errors and traces_
>
> Hey !
>
> [mabed@legiony520 /]$ sudo howdy add
> [sudo] Mot de passe de mabed :
> Data files have not been [...]

It looks like the fedora installer doesn't place the dlib-data directory in there. You can clone the repo and copy the folder /src/dlib-data to /usr/lib64/security/howdy and then run the install script for the time being. This should download the needed data files

boltgolt, 2019-03-26:
Ideally the folder should be included in the installer and the download script should be run automatically. @luyatshimbalanga is that possible for the fedora release?

luyatshimbalanga, 2019-04-02:
@boltgolt I'll take a look

m-ryals, 2019-04-07:
it may be helpful to include that in this situation, face id is using the data files which have been downloaded in all other situations, but it is only erroring with i3lock.

luyatshimbalanga, 2019-04-08:
@BedoM 
Could you test the updated Fedora version? The missing dlib-data should be installed.

BedoM, 2019-04-08:
Yes no problem

luyatshimbalanga, 2019-04-12:
Lets close this issue as the fix is in place.

boltgolt, 2019-04-12:
Thanks for the fix @luyatshimbalanga!

closed 2019-04-12 by boltgolt