boltgolt/howdy #175: Freenect2 Support

Stark003, 2019-03-23:
Support for Microsoft Kinect v2 IR

boltgolt, 2019-03-26:
Is the Kinect implemented as a V4L camera device? (does it show up in `v4l2-ctl --list-devices`?)

Stark003, 2019-03-28:
No, 

it responds with 
```
Failed to open /dev/video0: No such file or directory
```

I have libreenect2 installed and am able to access the various streams when using the premade examples that come with the program.
The kinect's video camera is also not recognised as a video device so I am also unable to use it as a webcam for my pc.

boltgolt, 2019-03-30:
In that case Kinect support will have to be written as a serperate recorder. I'm not going to be actively working on this, but if someone wants to implement it i will merge it.

closed 2019-05-08 by boltgolt