boltgolt/howdy #176: sudo in ssh session not working

ju916, 2019-03-28:
If I run sudo in an ssh session, authentication fails.

**Reproduce**:

1) Excecute "sudo -k id" -- it works with howdy
2) ssh to localhost and start "sudo -k id" -- it fails 

**The reason**: howdy is included as _required_ in _common-auth_
sudo includes common-auth (almost all authentication does!)
pam.py detects that it is called inside of a SSH connection (checking environment), decides, that face recognition can't work from remote and exits -- i.e. authentication fails.

**Warning**: Including a thing like howdy in _common-auth_ means asking for trouble. This file is supposed to be used by all kinds of applications. Including an auth scheme that is limited to certain environments will break Linux systems in ways we can't possibly imagine. And yes, this will also hurt unexperienced linux users sooner or later.

**Fix**: 

1) By default only include howdy only in selected pam configurations like gdm-password, gnome-screensaver, ... Provide assistance to include it in more areas

2) Maybe it makes sense to invest a little work in a more evolved PAM configuration. As an example: I am using howdy in a setup where the regular password login always works as a fallback for 2FA with a short PIN+howdy (common-howdy2fa):

```
# first factor
#auth    [success=ignore default=1]	pam_u2f.so
auth	[success=ignore default=1]	pam_userdb.so crypt=crypt db=/etc/pinlist
# second factor
auth	[success=2 default=ignore] 	pam_python.so /lib/security/howdy/pam.py

# Fallback -- common password
auth	[success=1 default=ignore]	pam_unix.so	nullok_secure try_first_pass 
# no success jump target: fail
auth	requisite			pam_deny.so
# Jump target: success 
auth    required                        pam_permit.so 
```

**Notes**: 
1) I replace "include common-auth" by "include common-howdy2fa" in selected pam configurations like sudo, xscreensaver, ... But with the fallback to regular password login you could think about using it as a replacement for common-auth.
2) _u2f_ uses simple security tokens for ~20 Euro (example yubikey). Working very well, very secure. Get one if you can.
3) _userdb_ is a standard unix PAM module, using a local database file. I use it to store username/PIN combinations (very nice to use for 2FA with howdy instead of a long password). I can provide a small script for creating the database file, if you want.

boltgolt, 2019-03-30:
Can't reproduce, on my machine PAM immediately falls back on the default unix password module:

[image]

Howdy hasn't been automatically writing to `common-auth` for a few months though, and instead switched to a [config file](https://github.com/boltgolt/howdy/blob/master/src/pam-config/howdy) in 2.5.0. This file should complete auth on success (`success=end`), but continue to other modules on failure (`default=ignore`)

I agree that the "required" keyword of the older versions was risky. If i recal correctly the only reason it was set to required in the first place was because PAM module examples did so, which is terrible coding

ju916, 2019-04-01:
You are right. In the current default configuration, this is not an issue. (I propably had an old version installed on my system and because I edited my pam files the updates didn't update this). Sorry for the fuzz.

I am closing this.

closed 2019-04-01 by ju916

boltgolt, 2019-04-01:
No problem!