boltgolt/howdy #177: Unable to log into my system. (Ubuntu, KDE

ujjwalbe, 2019-03-28:
I'm unable to log into my system, even I provide the password it Detetected my face earlier when I was in terminal and system was loged, but i can't log into my system anymore.  Please help as soon as possible.

boltgolt, 2019-03-28:
Did you change anything about the default PAM configuration?

ujjwalbe, 2019-03-28:
No I didn't change anything, and Don't even touch any config file.  Just installed it from the terminal. I think it's in a loop

boltgolt, 2019-03-28:
I think the quickest way to get back into your system is by using another Ubuntu or Windows instance, do you have a live USB stick?

boltgolt, 2019-03-28:
I've never seen it before, but i think @ju916 feels pretty vindicated right now. I'm guessing it's a combination of #139 and a non-default PAM setup. Good to hear you got access back.

Version 2.6.0 is planned for tomorrow and should include a fix for this.

ujjwalbe, 2019-03-28:
Yes i reads    @ju916 issue.
Currently i disabled it untill i get green signal from you. 
I was so happy when first time it unlocked without any issue and did sudo with my face that was awesome. This is a great work.

boltgolt, 2019-03-29:
Fixed in 2.5.1, as auto dismiss of the lockscreen has been removed

closed 2019-03-29 by boltgolt