boltgolt/howdy #179: How to trigger Howdy when unlocking sceen

jamy3531, 2019-03-31:
Here is how I do for now. Enter something (not necessary my login password) in Password Box first and Press the Return button, then Howdy starts to work.  I can't find another easier way.

Is there any shortcut to trigger Howdy  when need to unlock screen

boltgolt, 2019-04-01:
Howdy should be triggered before anything else. What are the "Priority" values in `/usr/share/pam-configs/unix` and `/usr/share/pam-configs/howdy`?

jamy3531, 2019-04-01:
The "Priority" is 256 in unix  and 512 in howdy. 

I think I may not describe my problem clearly. Howdy works file When login from a boot. But when I need unlock the screen after a privous lockscreen,  the IR camera starts to blink only after entering something in password box. So is there any shortcut，like pressing a button， to trigger the authentication process.

boltgolt, 2019-04-01:
Howdy should always be called before the unix password auth, so it might be crashing. Are there any errors in `tail /var/log/auth.log`?

CybernetiX-S3C, 2019-04-01:
How to prioritize howdy and unix? I am getting the same issue here. Also you will have to edit the pam.py file. ConfigParse was renamed to configparse. ``` import configparser

# Read config from disk
config = configparser.ConfigParser()
config.read(os.path.dirname(os.path.abspath(__file__)) + "/config.ini")

```

```
 tail /var/log/auth.log
Apr  1 10:25:01 CybernetiX-S3C CRON[11487]: pam_unix(cron:session): session opened for user root by (uid=0)
Apr  1 10:25:01 CybernetiX-S3C CRON[11487]: pam_unix(cron:session): session closed for user root
Apr  1 10:27:02 CybernetiX-S3C CRON[600]: pam_unix(cron:session): session opened for user daemon by (uid=0)
Apr  1 10:27:02 CybernetiX-S3C CRON[600]: pam_unix(cron:session): session closed for user daemon
Apr  1 10:30:01 CybernetiX-S3C CRON[724]: pam_unix(cron:session): session opened for user root by (uid=0)
Apr  1 10:30:01 CybernetiX-S3C CRON[724]: pam_unix(cron:session): session closed for user root
Apr  1 10:35:01 CybernetiX-S3C CRON[22309]: pam_unix(cron:session): session opened for user root by (uid=0)
Apr  1 10:35:01 CybernetiX-S3C CRON[22309]: pam_unix(cron:session): session closed for user root
Apr  1 10:39:01 CybernetiX-S3C CRON[806]: pam_unix(cron:session): session opened for user root by (uid=0)
Apr  1 10:39:01 CybernetiX-S3C CRON[806]: pam_unix(cron:session): session closed for user root
```

boltgolt, 2019-04-01:
ConfigParser was renamed to configparser in python3, but `pam-python` still runs in python2. It should therefore still be called as "ConfigParser" in `pam.py`.

Your auth log looks fine, and this issue was never reported under Howdy <2.5.0. Might have been a a change in the last release that accidentally broke this.

jamy3531, 2019-04-02:
> Howdy should always be called before the unix password auth, so it might be crashing. Are there any errors in `tail /var/log/auth.log`?

Strangely, theres's no /var/log/auth.log file in my system. My distribution is Deepin.

sthpravin, 2019-04-26:
Hi, @boltgolt, Face Authentication works fine while booting up ubuntu 16. It also works when when authenticating from terminals. But whenever I log out of the system and try to log in, it shows displays unknown error 1 and authentication does not work

boltgolt, 2019-04-28:
Does `tail /var/log/auth.log` output anything for you after triggering the error?

viewv, 2019-05-21:
> > Howdy should always be called before the unix password auth, so it might be crashing. Are there any errors in `tail /var/log/auth.log`?
> 
> Strangely, theres's no /var/log/auth.log file in my system. My distribution is Deepin.

I am using deepin too, their is no file named auth.log under /var/log/. Every time  I need to unlock my computer, I have to input my password.

papermache, 2019-11-14:
I've found Howdy is better at login than unlock so if you need quick access restart your window session.

dietmargerlach, 2020-03-14:
Same issue for me:
Output from auth.log after suspend and trying to log back in:
Mar 14 17:21:58 didi-TUXEDO systemd-logind[866]: Suspend key pressed.
Mar 14 17:22:06 didi-TUXEDO dbus-daemon[885]: [system] Rejected send message, 1 matched rules; type="error", sender=":1.93" (uid=1000 pid=2726 comm="/usr/bin/pulseaudio --start --log-target=syslog " label="unconfined") interface="(unset)" member="(unset)" error name="org.bluez.MediaEndpoint1.Error.NotImplemented" requested_reply="0" destination=":1.29" (uid=0 pid=1357 comm="/usr/lib/bluetooth/bluetoothd " label="unconfined")
Mar 14 17:22:06 didi-TUXEDO dbus-daemon[885]: message repeated 3 times: [ [system] Rejected send message, 1 matched rules; type="error", sender=":1.93" (uid=1000 pid=2726 comm="/usr/bin/pulseaudio --start --log-target=syslog " label="unconfined") interface="(unset)" member="(unset)" error name="org.bluez.MediaEndpoint1.Error.NotImplemented" requested_reply="0" destination=":1.29" (uid=0 pid=1357 comm="/usr/lib/bluetooth/bluetoothd " label="unconfined")]
Mar 14 17:22:07 didi-TUXEDO systemd-logind[866]: Operation 'sleep' finished.
Mar 14 17:22:09 didi-TUXEDO /lib/security/howdy/pam.py[19349]: Traceback (most recent call last):
Mar 14 17:22:09 didi-TUXEDO /lib/security/howdy/pam.py[19349]:   File "/lib/security/howdy/pam.py", line 68, in pam_sm_authenticate
Mar 14 17:22:09 didi-TUXEDO /lib/security/howdy/pam.py[19349]:     return doAuth(pamh)
Mar 14 17:22:09 didi-TUXEDO /lib/security/howdy/pam.py[19349]:   File "/lib/security/howdy/pam.py", line 62, in doAuth
Mar 14 17:22:09 didi-TUXEDO /lib/security/howdy/pam.py[19349]:     pamh.conversation(pamh.Message(pamh.PAM_ERROR_MSG, "Unknown error: " + str(status)))
Mar 14 17:22:09 didi-TUXEDO /lib/security/howdy/pam.py[19349]: PamException: Kommunikationsfehler
Mar 14 17:22:11 didi-TUXEDO sshd[1126]: Received SIGHUP; restarting.
Mar 14 17:22:11 didi-TUXEDO sshd[1126]: Server listening on 0.0.0.0 port 22.
Mar 14 17:22:11 didi-TUXEDO sshd[1126]: Server listening on :: port 22.
Mar 14 17:22:11 didi-TUXEDO sshd[1126]: Received SIGHUP; restarting.
Mar 14 17:22:11 didi-TUXEDO sshd[1126]: Server listening on 0.0.0.0 port 22.
Mar 14 17:22:11 didi-TUXEDO sshd[1126]: Server listening on :: port 22.
Mar 14 17:22:18 didi-TUXEDO unix_chkpwd[19565]: password check failed for user (didi)
Mar 14 17:22:18 didi-TUXEDO gnome-screensaver-dialog[19349]: pam_unix(gnome-screensaver:auth): authentication failure; logname= uid=1000 euid=1000 tty=:0.0 ruser= rhost=  user=didi
Mar 14 17:22:22 didi-TUXEDO /lib/security/howdy/pam.py[19349]: Traceback (most recent call last):
Mar 14 17:22:22 didi-TUXEDO /lib/security/howdy/pam.py[19349]:   File "/lib/security/howdy/pam.py", line 68, in pam_sm_authenticate
Mar 14 17:22:22 didi-TUXEDO /lib/security/howdy/pam.py[19349]:     return doAuth(pamh)
Mar 14 17:22:22 didi-TUXEDO /lib/security/howdy/pam.py[19349]:   File "/lib/security/howdy/pam.py", line 62, in doAuth
Mar 14 17:22:22 didi-TUXEDO /lib/security/howdy/pam.py[19349]:     pamh.conversation(pamh.Message(pamh.PAM_ERROR_MSG, "Unknown error: " + str(status)))
Mar 14 17:22:22 didi-TUXEDO /lib/security/howdy/pam.py[19349]: PamException: Kommunikationsfehler
Mar 14 17:22:30 didi-TUXEDO gnome-screensaver-dialog[19349]: gkr-pam: unlocked login keyring

closed 2020-09-28 by boltgolt