boltgolt/howdy #181: Currently Not working with sddm

CybernetiX-S3C, 2019-04-01:
_Please describe the issue in as much detail as possible, including any errors and traces_
I have everything installed properly. Tests works. Although the IR light nor howdy in general is working on the SDDM greeter or the SDDM lock screen.

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Kali Linux 2019.1, KDE, SDDM, 4.19

Howdy version: 2.5.1

boltgolt, 2019-04-11:
I'm not 100% sure what the issue is here, especially if you're sure dlib is installed correctly. I've seen some users fix this by calling pip install as a subprocess within `compare.py` but i wouldn't recommend that.

kabob8933, 2019-10-26:
I think this has something todo with SDDM not behaving with PAM, as i was having this issue, i switched to GDM, and it worked.

boltgolt, 2019-11-11:
If SDDM is to blame then i can't do anything about it, but there's not a lot of activity on this issue so it might be a small problem

boltgolt, 2020-01-07:
Closing due to inactivity

closed 2020-01-07 by boltgolt

DavidM42, 2020-01-28:
I also have problems with sddm. I have pop OS installed and it worked with it's default gdm and gnome 3 configuration. I installed KDE for perfomance reasons. Howdy auth in terminal still works and on booting the system the first login via gdm lock screen works. On any subsequent screen locks and unlocks (with kde lock screen) I can only use passwords to login in sddm.
I tested [these tips](https://github.com/boltgolt/howdy/issues/219#issuecomment-502054807) but with these lines added I can't login at all in the sddm/kde lockscreen

MarkoShiva, 2020-01-29:
@boltgolt @DavidM42 It works on my machine. :D 
Well in case of sddm you have two files in `/etc/pam.d/` sddm and sddm-greater that you need to edit and add `auth sufficient pam_python.so /lib/security/howdy/pam.py` 
Then once you logout and try to login all you have to do is press > button for login on sddm or click enter it depends on theme I think. 
But once you trigger the password verification the camera will light up and if you have setup face recognition you can log in with that.

CybernetiX-S3C, 2020-02-01:
I have done all the recommendations here, even prior to my issue post. I
have also done the recommendations for the import cv2 error. All while I
have everything installed correctly.

John Poli Modica
Cyber Security Professional
CybernetiX S3C
https://CybernetiX-S3C.net/

CybernetiX-S3C, 2020-02-01:
I'm still having an issue.

John Poli Modica
Cyber Security Professional
CybernetiX S3C
https://CybernetiX-S3C.net/

davi-ro, 2020-03-05:
Try these steps maybe it will help, had same problems in kde neon
https://github.com/boltgolt/howdy/issues/302#issuecomment-595167414

iamchapita, 2020-09-02:
> Try these steps maybe it will help, had same problems in kde neon
> [#302 (comment)](https://github.com/boltgolt/howdy/issues/302#issuecomment-595167414)

This works for me in KDE Neon in loggin Screen.