boltgolt/howdy #182: issues with lock scripts like i3lock, slock, etc

m-ryals, 2019-04-02:
so i've noticed that scripts like i3lock dont activate the camera until after a password is entered. even then, the module fails with this error:
Data files have not been downloaded, please run the following commands:

	cd /lib/security/howdy/dlib-data
	sudo ./install.sh

Traceback (most recent call last):
  File "/lib/security/howdy/compare.py", line 219, in <module>
    face_locations = face_detector(gsframe, 1)
TypeError: 'NoneType' object is not callable
Data files have not been downloaded, please run the following commands:

	cd /lib/security/howdy/dlib-data
	sudo ./install.sh

Traceback (most recent call last):
  File "/lib/security/howdy/compare.py", line 219, in <module>
    face_locations = face_detector(gsframe, 1)
TypeError: 'NoneType' object is not callable

However, it works in all other applications, and I do have the data files downloaded. i'm not sure what is going wrong, or if this is somethibng that is related to my config, but if you could please breifly explain what is happening and why, that would literally make my day. thanks!

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): ubuntu 18.10

Howdy version: latest as of april 1st 2019

boltgolt, 2019-04-02:
Looks like `PATH` is not pointing to the right folder. Could you add the following statement to `compare.py` on line 26: (don't forget the tab)

```python
	print(PATH)
```

What does it print in the error?

m-ryals, 2019-04-02:
```
/lib/security/howdy
Data files have not been downloaded, please run the following commands:

	cd /lib/security/howdy/dlib-data
	sudo ./install.sh

Traceback (most recent call last):
  File "/lib/security/howdy/compare.py", line 219, in <module>
    face_locations = face_detector(gsframe, 1)
TypeError: 'NoneType' object is not callable
```
Its the same error. the files are there, but it just is giving that error.

boltgolt, 2019-04-07:
This is the same issue as discussed [here](https://github.com/boltgolt/howdy/issues/174#issuecomment-476442439), closing this for now to keep the discussion in 1 place.

closed 2019-04-07 by boltgolt

pthorin, 2019-09-09:
I don't think this is the same issue as #174, I still have this issue in Ubuntu 18.03 using howdy 2.5.1. 

The dlib-data files are in `/lib/security/howdy/dlib-data` and howdy works fine for sudo, logging in using gdm, unlocking screenlock in Gnome, but not for i3lock or slock. 

Could it be some kind of permission issue?

pthorin, 2019-09-09:
The `dlib-data` directory had root:root 0744 permissions, meaning that a normal user cannot access any of the files in the directory, except seeing that they are there, even though the files themselves are readable by any user. 
Changing the permissions to 0755 fixes the problem. 

I don't know what the security implications would be for doing this?