boltgolt/howdy #183: Fedora 29 Gnome login not working

kervin5, 2019-04-02:
Hi, 

Thanks a lot for this awesome package. I managed to get it working for sudo and other prompts in the system(Fedora 29) but for some reason it doesn't work in the login screen. I shows a brief message that says "attempting face detection" but it fails right away and says: Unknow error 1. I tried changing the permissions of the howdy folder but same result. When this occurs the camera/IR emitters don't turn on as it usually does when I use the sudo command in the terminal.
 
----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Fedora 29

Howdy version: 2.5.0

Germanc, 2019-04-04:
I have the same issue with 2.5.1 on Ubuntu 18.04 (KDE). It works with sudo but I can't login or unlock the screen, with error "Unknown error 1". The camera does turn on though for 4 seconds after I press enter

boltgolt, 2019-04-05:
Could you both try [this](https://github.com/boltgolt/howdy/issues/28#issuecomment-396231042) solution?

Germanc, 2019-04-05:
It worked on my computer. Thank you!

closed 2019-04-07 by boltgolt