boltgolt/howdy #185: Unable to install for kali linux 2019.1

spdhiraj99, 2019-04-11:
`Dependency is not satisfiable: streamer`
I get this error when trying to install howdy. 
i tried to install streamer using `apt-get install streamer` 
`Package streamer is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
E: Package 'streamer' has no installation candidate
`
seems like it's not available on kali repo.
so is there any workaround for this? 

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Kali linux 2019.1

Howdy version:2.5.1

boltgolt, 2019-04-11:
Streamer is [only used to automatically fire webcams for IR camera detection](https://github.com/boltgolt/howdy/blob/master/debian/preinst#L76). You can remove the dependency and use the `HOWDY_NO_PROMPT` flag to skip webcam detection. Be sure to manually set the `device_path` config variable to the location of your IR camera.

spdhiraj99, 2019-04-11:
oh okay, I'll try doing that. And can you tell me how to install it after changing the script? do i have to run preinst, install and postint ?

boltgolt, 2019-04-11:
Installing the .deb from the releases page with `dpkg` will not install dependencies, but still runs the installation scripts. If you run `HOWDY_NO_PROMPT=1` before installation you should not be prompted for webcam detection.

spdhiraj99, 2019-04-12:
`dpkg: error processing archive howdy-master.deb (--install):
 unable to open file '/var/lib/dpkg/tmp.ci//source': Is a directory
Errors were encountered while processing:` I'm getting this error now

denics, 2019-05-05:
Hi, I can confirm the same error with your release package 2.5.1.deb and Debian Buster:
```
This package is uninstallable
Dependency is not satisfiable: streamer
```

boltgolt, 2019-05-08:
A workaround for the streamer error is posted above, but the "Is a directory" error is more concerning. Is that error also showing up for you @denics?

denics, 2019-05-11:
I'm on Debian and I do not get the "directory" error. Also in Debian the install via dpkg does not work:
``HOWDY_NO_PROMPT=1 sudo -E dpkg -i howdy_2.5.1.deb``
do you think it would be possible to have a 2.5.1_1 without the **streamer** dependency?

boltgolt, 2019-05-16:
I'd have to rewrite the auto detection code to drop streamer, it's on the todo list

DonGatoMalo, 2019-05-16:
Debian Buster RC1 . The same issue.

b00f, 2019-08-26:
Thanks for this library. I used it on Ubuntu, However I can't install  it on debian 10 (buster). Hopefully you can fix it soon.

stannynuytkens, 2019-10-18:
I need this! :yum:

boltgolt, 2020-06-22:
Streamer is now optional, should fix this

closed 2020-06-22 by boltgolt