boltgolt/howdy #188: Dell 9370 Issues

emmertex, 2019-04-12:
I've gone through all of the possible fixes with no success, but in the end I have had it work.

The only change to the code needed was the array size.

`Traceback (most recent call last):
  File "/usr/local/bin/howdy", line 89, in <module>
    import cli.add
  File "/lib/security/howdy/cli/add.py", line 129, in <module>
    video_capture.grab()
  File "/lib/security/howdy/recorders/pyv4l2_reader.py", line 74, in grab
    self.read()
  File "/lib/security/howdy/recorders/pyv4l2_reader.py", line 92, in read
    img2 = (numpy.frombuffer(img_bgr, numpy.uint8).reshape([352, 352, 3]))
ValueError: cannot reshape array of size 346800 into shape (352,352,3)
`
I changed the array to what it needed to be based on the array size, 340, 340, 3
and it functioned.

I could not get it to work using opencv or ffmpeg at all.

These are the settings in config - wether these are all needed I am not sure, but device mode of anything else did not work.
`
# Set the camera input profile to this width and height
# The largest profile will be used if set to -1
# Automatically ignored if not a valid profile
frame_width = 340
frame_height = 340

# Because of flashing IR emitters, some frames can be completely unlit
# Skip the frame if the lowest 1/8 of the histogram is above this percentage
# of the total
# The lower this setting is, the more dark frames are ignored
dark_threshold = 50

# The recorder to use. Can be either opencv (default), ffmpeg or pyv4l2.
# Switching from the default opencv to ffmpeg can help with grayscale issues.
recording_plugin = pyv4l2
# opencv

# Video format used by ffmpeg. Options include vfwcap or v4l2.
# FFMPEG only.
device_format = v4l2

# Force the use of Motion JPEG when decoding frames, fixes issues with YUYV
# raw frame decoding.
# OPENCV only.
force_mjpeg = false
`

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable):
Distributor ID: Ubuntu
Description:    Ubuntu 18.10
Release:        18.10
Codename:       cosmic
4.18.0-17-generic

Howdy version:
2.5.1 deb release

emmertex, 2019-04-14:
I am now running Ubuntu 19.04
For whatever pyv4l2 will not build, changing the recorder to v4l2 solved that issue.

I tested with and without the array modification above and both work as expected.

ffmpeg (AttributeError: module 'ffmpeg' has no attribute 'input')
opencv also works now.

Distributor ID: Ubuntu
Description:    Ubuntu Disco Dingo (development branch)
Release:        19.04
Codename:       disco

AnthonyWharton, 2019-05-15:
@emmertex `pyv4l2` appears to be broken at the moment, we've been discussing this [over here](https://github.com/boltgolt/howdy/issues/160#issuecomment-492265683).  Your best bet would probably be to avoid this and wait for the issue to be fixed over there (i[t's been reported](https://github.com/duanhongyi/pyv4l2/issues/3)).

I think that [Line 92](https://github.com/boltgolt/howdy/blob/master/src/recorders/pyv4l2_reader.py#L92) is mistakenly hardcoded, instead of `352x352` it should use the `height` and `width` fields stored within the class (extracted from the `probe()` method). 

I'd have a go at changing it and making a PR if that were the case, but I cannot test and reproduce this without the pip dependency working. @boltgolt can you confirm this?

mdnahas, 2020-01-10:
I wasn't sure where to leave my experience.  Maybe there's a place for it on the wiki?

I've got a Dell XPS 13 9370.  Running Ubuntu 19.10.  

I ran:
```
  sudo add-apt-repository ppa:boltgolt/howdy
  sudo apt update
  sudo apt install howdy
```

I got this error:
```
>>> Installing OpenCV

Traceback (most recent call last):
  File "/usr/bin/pip3", line 11, in <module>
    sys.exit(main())
TypeError: 'module' object is not callable
Error while running last command
```

The version of pip3 that ships with Ubuntu 19.10 is outdated.  I used [this fix for pip3](https://github.com/boltgolt/howdy/issues/265#issuecomment-545704230).  

After fixing pip3, I ran:
```
  sudo apt remove howdy
  sudo apt autoremove
  sudo apt install howdy
```
Ubuntu popped up crash reports on pip3 and the howdy install process.

The command `sudo howdy add` seemed to work.

To test, I needed to run `sudo QT_X11_NO_MITSHM=1 howdy test`.  It blinks a lot.  It puts a circle around my face and not around my coffee cup, so it looked like it was recognizing me.

I was able to login using my face and use `sudo` with my face.  I don't know what happened during the install process, but it seems to be working.

FYI, /dev/video0 is the color webcam and /dev/video2 is the IR camera.

boltgolt, 2020-09-28:
Closing this as it has gone very stale

closed 2020-09-28 by boltgolt