boltgolt/howdy #191: What is the principle of howdy?

bingws, 2019-04-19:
What is the principle of howdy? Do you have any introductory documents? Mainly about how to interact with the Linux kernel @boltgolt

boltgolt, 2019-04-19:
There's a slightly outdated [medium post](https://medium.com/@boltgolt/hacking-together-windows-hello-on-ubuntu-ecd97ff16430) that explains the basis of Howdy, would that answer your question?

bingws, 2019-04-19:
@boltgolt The URL is walled，no wonder it can't be seen. I've seen it before.

bingws, 2019-04-19:
[image]
What about these values:10,11,12? How to calculate them?

boltgolt, 2019-04-19:
The medium article is not behind a paywall as far as i can tell.

They're not calculated values, they're status codes. The meaning of the codes is shown in the comments above the if statements.

You do not need to mention me in every comment, i respond when i can.

bingws, 2019-04-19:
Why is that status code not 7、8、 9，but is  10, 11, 12?

boltgolt, 2019-04-19:
That's just how i wrote them

bingws, 2019-04-19:
So I can write 7, 8, 9, and that's OK? How to get the status code？

bingws, 2019-04-22:
I found "python-pam" support python3 now, https://pypi.org/project/python-pam/

bingws, 2019-04-23:
I found that these status codes are not reliable. The program will not have an error like "Face detection timeout reached". Instead, I directly reported "Unknown error: 1" like following on Ubuntu16.4:
[image]

boltgolt, 2019-04-23:
No sure what you're trying to achieve, but error 1 is a python crash. Basically all programs exit with code 1 on error. The >10 status codes Howdy uses are reliable.

I'm closing this issue now as the original question has been answered, and multiple off-topic comments have been posted and subsequently deleted.

closed 2019-04-23 by boltgolt