boltgolt/howdy #197: face_detector() throws an exception when used in compare.py, line 19

Ethiarpus, 2019-04-28:
This began happening when I updated to Howdy 2.5.1.

Error thrown by Python:
```
Traceback (most recent call last):
  File "/lib/security/howdy/compare.py", line 219, in <module>
    face_locations = face_detector(gsframe, 1)
RuntimeError: Unsupported image type, must be 8bit gray or RGB image.
Unknown error: 1

```

For reference, face_detector is
`<dlib.fhog_object_detector object at 0x7f3e1574cc38>`
And gsframe is
```
[[63 63 63 ... 58 58 58]
 [63 63 63 ... 58 58 58]
 [63 63 63 ... 58 58 58]
 ...
 [91 91 91 ... 83 83 83]
 [91 91 91 ... 83 83 83]
 [91 91 91 ... 83 83 83]]

```
Other potentially useful info:
`use_cnn` is false

I don't know where the bug template went but here's some system information
OS: Mint 18.3 sylvia (debian-based OS btw)
Kernel: x86_64 Linux 4.15.0-45-generic
Shell: bash 4.3.48
Resolution: 1920x1080
Cinnamon 3.6.7
WM: Muffin
Howdy Version: 2.5.1

Ethiarpus, 2019-05-01:
@boltgolt ?

boltgolt, 2019-05-08:
What does 

```
v4l2-ctl -d /dev/video0 --list-formats
```
and

```
dmesg | grep "Unknown video format"
```

Output?

saurabhdhillon, 2019-05-13:
I am using ubuntu 16.04 LTE, and facing "Unknown error: 1". At login when i press enter with no password, cam turns on and message is displayed "Unknown error: 1".

Output for : "v4l2-ctl -d /dev/video0 --list-formats"

```
ioctl: VIDIOC_ENUM_FMT
	Index       : 0
	Type        : Video Capture
	Pixel Format: 'MJPG' (compressed)
	Name        : Motion-JPEG

	Index       : 1
	Type        : Video Capture
	Pixel Format: 'YUYV'
	Name        : YUYV 4:2:2
```

Nothing for second command.

saurabhdhillon, 2019-05-14:
@boltgolt your views on this?

boltgolt, 2019-05-14:
@saurabhdhillon that's no the same problem as discussed in this issue. Please search for another issue that does match your problem.

saurabhdhillon, 2019-05-15:
@boltgolt will post this as new issue.

Ethiarpus, 2019-05-15:
```bash
ethiarpus@ethiarpus ~ $ v4l2-ctl -d /dev/video0 --list-formats
ioctl: VIDIOC_ENUM_FMT
	Index       : 0
	Type        : Video Capture
	Pixel Format: 'MJPG' (compressed)
	Name        : Motion-JPEG

	Index       : 1
	Type        : Video Capture
	Pixel Format: 'YUYV'
	Name        : YUYV 4:2:2

ethiarpus@ethiarpus ~ $ dmesg | grep "Unknown video format"
```
The dmesg gave no output, even after running `sudo test` to get the cameras to fire

Ethiarpus, 2019-05-16:
I tried `apt uninstall`ing Howdy and `apt autoremove`ing everything else. Then I re-`apt install`ed Howdy and everything was well.

closed 2019-05-16 by Ethiarpus

saurabhdhillon, 2019-05-16:
@Ethiarpus can you help me out for issue [#212](https://github.com/boltgolt/howdy/issues/212)