boltgolt/howdy #201: Howdy for root login

aker99, 2019-05-07:
Whenever try to add a module for `root` user it gives the following error
*Command:* `sudo howdy add --user root`
`Can't run howdy commands as root, please run this command with the --user flag`
It would great to add feature so that we can also add module for root account, as while working in a non root it's very tedious task to keeping entering  password for root.

boltgolt, 2019-05-08:
A root user should not have to authenticate with a PAM module normally, as he has the highest privileges already. I won't be working on this, unfortunately. However, you could just sun `sudo -i` for a persistent root shell.

closed 2019-05-08 by boltgolt

dsvf, 2023-06-02:
There is a valid use case for this - when sudo is configured with `rootpw`, it requires the _root_ password to authenticate as a user. In that case, sudo doesn't work with howdy.