boltgolt/howdy #203: Port Howdy to OpenSuse

dmafanasyev, 2019-05-07:
Would it be possible to port Howdy to OpenSuse (Leap 15.0/1 or Tumbleweed) and build rpm?

I have checked the offical Tumbleweed repos for some dependencies of Howdy (based on https://github.com/boltgolt/howdy/blob/master/debian/control#L12 and https://github.com/boltgolt/howdy/blob/master/fedora/howdy.spec) and can not find only one package:

polkit-devel - ok
curl|wget - ok
python3 - ok
python3-base - ok
python3-pip - ok
python3-dev - ok, but named python3-devel
python3-setuptools - ok
python3-opencv - ok
libpam-python - ok, but named pam-python or python3-python-pam
fswebcam - ok
libopencv-dev - ok, but named opencv-devel (there is libopencv3_4 package also)
cmake - ok
streamer - ok, if this is the GStreamer (there are gstreamer, gstreamer-plugin-python, python3-gst packages also)
libv4l2-0  - ok
python3-face_recognition - no

Thanks in advance!

boltgolt, 2019-05-08:
It should be possible, especially if you consider that `python3-face_recognition` is an unused dependency at this point, and that `streamer` is only used to detect the right camera and not necessary at all. Watch out for packages called `python-pam` though, it usually does the exact opposite of what Howdy needs (start a PAM auth from a python script).

I'm quite unfamiliar with the distro, but anyone is free to create a port if they want to (as happened with the arch and fedora ports).

dmafanasyev, 2019-06-06:
Hi @boltgolt ,

Thanks a lot for your answer.

I have started from Fedora rpm .spec file of @luyatshimbalanga and build howdy rpm package for openSUSE. This wiki page describe installation and configuration steps in details:

[https://en.opensuse.org/SDB:Facial_authentication](url)

It may be useful to place this link into README.md.

dmafanasyev, 2019-06-07:
Some additional info.

Here you csn find howdy.spec file for rpm package:

https://build.opensuse.org/package/view_file/home:dmafanasyev/howdy/howdy.spec?expand=1

Also I have packaged dlib because I could not found it for openSUSE in standard or community repo (howdy rpm package depend on python-dlib):

https://build.opensuse.org/package/show/home:dmafanasyev/python-dlib

boltgolt, 2019-08-02:
Sorry for not getting back to you sooner, amazing work!

I've added openSUSE as a supported platform in the readme, see #240. Your write-up on the wiki is great and easy to follow, thanks a lot!

closed 2019-08-02 by boltgolt