boltgolt/howdy #208: Unable to login on Kubuntu 19.04 (KDE 5.15 SDDM)

mijorus, 2019-05-13:
Already found issue #177 about a similar problem, however I am running the latest version and the same problem is also present. Futhermore, Howdy seems to break the login compleatly and it keeps me out even when i enter the correct password. I need to accest the terminal with ctrl+alt+F4 and disable Howdy. I found that it only happens after a reboot, not when i try to login after it went to sleep

boltgolt, 2019-08-02:
Do you have disk encryption enabled by any chance?

orclex, 2019-09-25:
Hi,

I cannot login with howdy on KDE loginmanager SDDM.  The login screen turns grey and gets stuck!
If howdy is enabled I also cannot login with my password in SDDM - it gets stuck too.

In general howdy is working. I can switch to the terminal with ctrl+alt+F3, enter my username and I will be logged in automatically by howdy face authentication. sudo is working too.

My homedirectory is encrypted. I'm running Kubuntu 19.04 with KDE 16.5.
Is there any workaround or fix available?

Thanks!

boltgolt, 2019-09-25:
Could be a folder permissions issue, search for it in the issues here

orclex, 2019-09-25:
hm, any hints where to search? Where does howdy needs which permissions?

orclex, 2019-09-25:
ah, found the issue in bug reports and will check it...

orclex, 2019-09-25:
Unfortunately it is not working:
I tried setting permissions
`chmod o+x /lib/security/howdy/dlib-data`
from  **Failing to authenticate on lockscreen #238**  but SDDM still gets stuck. Permissions for dlib-data are executable for all.

Any ideas?

orclex, 2019-09-26:
Permissions are set to:

```
18620854      4 drwxr-xr-x   3 root     root         4096 Sep 26 08:06 /lib/security/
 19144864      4 drwxr-xr-x   7 root     root         4096 Sep 26 07:37 /lib/security/howdy
 19144866      4 drwxr-xr-x   3 root     root         4096 Sep 25 09:36 /lib/security/howdy/cli
 19144884      4 drwxr-xr-x   2 root     root         4096 Sep 25 11:00 /lib/security/howdy/cli/__pycache__
 19145417      4 -rw-r--r--   1 root     root         1027 Sep 25 11:00 /lib/security/howdy/cli/__pycache__/disable.cpython-37.pyc
 19145414      4 -rw-r--r--   1 root     root         3932 Sep 25 10:24 /lib/security/howdy/cli/__pycache__/test.cpython-37.pyc
 19145409      4 -rw-r--r--   1 root     root         1020 Sep 25 10:24 /lib/security/howdy/cli/__pycache__/list.cpython-37.pyc
 19145413      4 -rw-r--r--   1 root     root          434 Sep 25 10:14 /lib/security/howdy/cli/__pycache__/config.cpython-37.pyc
 19145412      4 -rw-r--r--   1 root     root         3719 Sep 25 09:36 /lib/security/howdy/cli/__pycache__/add.cpython-37.pyc
 19145411      4 -rw-r--r--   1 root     root          126 Sep 25 09:36 /lib/security/howdy/cli/__pycache__/__init__.cpython-37.pyc
 19145416      4 -rw-r--r--   1 root     root          779 Sep 25 10:25 /lib/security/howdy/cli/__pycache__/clear.cpython-37.pyc
 19144871      4 -rwxr-xr-x   1 root     root           34 Apr 14  2018 /lib/security/howdy/cli/__init__.py
 19144869      8 -rwxr-xr-x   1 root     root         6118 Jan  6  2019 /lib/security/howdy/cli/add.py
 19144874      4 -rwxr-xr-x   1 root     root          557 Jan  6  2019 /lib/security/howdy/cli/config.py
 19144868      4 -rwxr-xr-x   1 root     root         2431 Jan  6  2019 /lib/security/howdy/cli/remove.py
 19144870      4 -rwxr-xr-x   1 root     root         1004 Apr 14  2018 /lib/security/howdy/cli/clear.py
 19144867      8 -rwxr-xr-x   1 root     root         6737 Mär 29  2019 /lib/security/howdy/cli/test.py
 19144872      4 -rwxr-xr-x   1 root     root         1474 Jan  6  2019 /lib/security/howdy/cli/disable.py
 19144873      4 -rwxr-xr-x   1 root     root         1360 Jan  6  2019 /lib/security/howdy/cli/list.py
 19144881      4 -rwxr-xr-x   1 root     root         3026 Jan  6  2019 /lib/security/howdy/cli.py
 19144880      4 -rwxr--r--   1 root     root         2787 Mär 29  2019 /lib/security/howdy/pam.py
 19145415      4 -rwxr--r--   1 root     root         2488 Sep 26 07:37 /lib/security/howdy/config.ini
 19273564      4 drwxr-xr-x   2 root     root         4096 Sep 25 10:25 /lib/security/howdy/models
 19273565     12 -rw-r--r--   1 root     root         8441 Sep 25 10:25 /lib/security/howdy/models/alex.dat
 19144875      4 drwxr--r--   2 root     root         4096 Sep 25 09:32 /lib/security/howdy/recorders
 19144878     52 -rwxr--r--   1 root     root        52253 Jan  6  2019 /lib/security/howdy/recorders/v4l2.py
 19144876      8 -rwxr--r--   1 root     root         4700 Jan  6  2019 /lib/security/howdy/recorders/ffmpeg_reader.py
 19144877      0 -rwxr--r--   1 root     root            0 Jan  6  2019 /lib/security/howdy/recorders/__init__.py
 19144879      4 -rwxr--r--   1 root     root         2900 Mär 29  2019 /lib/security/howdy/recorders/pyv4l2_reader.py
 19144882      4 drwxr--r--   2 root     root         4096 Sep 25 09:32 /lib/security/howdy/pam-config
 19144883      4 -rwxr--r--   1 root     root          142 Jan  6  2019 /lib/security/howdy/pam-config/howdy
 19144885      4 drwxr--r-x   2 root     root         4096 Sep 25 09:32 /lib/security/howdy/dlib-data
 19144886      4 -rwxr--r--   1 root     root          412 Jan  6  2019 /lib/security/howdy/dlib-data/Readme.md
 19145410  21940 -rwxr--r--   1 root     root     22466066 Sep 25 09:32 /lib/security/howdy/dlib-data/dlib_face_recognition_resnet_model_v1.dat
 19144887      4 -rwxr--r--   1 root     root         1310 Jan  6  2019 /lib/security/howdy/dlib-data/install.sh
 19145408   8940 -rwxr--r--   1 root     root      9150489 Sep 25 09:32 /lib/security/howdy/dlib-data/shape_predictor_5_face_landmarks.dat
 19145407    716 -rwxr--r--   1 root     root       729940 Sep 25 09:32 /lib/security/howdy/dlib-data/mmod_human_face_detector.dat
 19144865     12 -rwxr--r--   1 root     root         9052 Mär 29  2019 /lib/security/howdy/compare.py
 18620856     20 -rw-r--r--   1 root     root        18680 Sep 25  2017 /lib/security/pam_ecryptfs.so
```

Executable permissions for everyone and all files with
`chmod -R a+x /lib/security/howdy`
is still not working.

Do you get howdy running on KDE SDDM loginmanager?

orclex, 2019-10-07:
any news on this? could you reproduce it?

Slaninecka, 2019-10-07:
I propably have the same issue, when i try to login with howdy enabled on my kubuntu 19.04 after I try to login my IR emmitters flash for a second and the login screen freezes. Changing permissions didnt help.

orclex, 2019-10-18:
Unfortunately howdy is still not working with KDE Plasma 17.0 on Kubuntu 19.10. Login via sddm (0.18.0) gets stuck while login via sudo works.
Could you please check what the problem is?

boltgolt, 2019-10-18:
Could you try setting `no_confirmation` to true in the config?

Slaninecka, 2019-10-18:
That worked for me, thanks.

orclex, 2019-10-18:
Unfortunately it is not working for me with 
`no_confirmation = true`
What else did you change? I'm still stuck in sddm after trying to login...

Slaninecka, 2019-10-19:
This is my config, other then that I only I changed the same permission settings as you said you did.
```
detection_notice = false
no_confirmation = true
suppress_unknown = false
ignore_ssh = true
ignore_closed_lid = true
disabled = false
use_cnn = false
certainty = 3
timeout = 3
max_height = 360
frame_width = -1
frame_height = -1
dark_threshold = 50
recording_plugin = opencv
device_format = v4l2
force_mjpeg = false
exposure = -1 
end_report = false
```

orclex, 2019-10-21:
My config is nearly the same but still not working. Which distribution do you use and which KDE plasma?

```
[core]
detection_notice = false
no_confirmation = false
suppress_unknown = false
ignore_ssh = true
ignore_closed_lid = true
disabled = true
use_cnn = false

[video]
certainty = 3.5
timeout = 4
device_path = /dev/video0
max_height = 320
frame_width = -1
frame_height = -1
dark_threshold = 50
recording_plugin = opencv
device_format = v4l2
force_mjpeg = false
exposure = -1

[debug]
end_report = false
```

Maybe **boltgolt** has some further ideas?

boltgolt, 2019-10-21:
Looks like your `no_confirmation` is still set to `false`?

Slaninecka, 2019-10-22:
```
Operating System: Kubuntu 19.04
KDE Plasma Version: 5.15.4
KDE Frameworks Version: 5.56.0
Qt Version: 5.12.2
Kernel Version: 5.3.2-050302-generic
OS Type: 64-bit
```

orclex, 2019-10-28:
Of course `no_confirmation` was set to `true` when I tried to login via howdy.

Anyway, I did some more tests...

In general login with howdy and sddm is working for common users if `no_confirmation` is set to `true`. 
But I experienced problems with encrypted home directories! My home is encrypted with `ecryptfs` and it should be decrypted on login. sddm still gets stuck and does not return input control to the user after trying to login with howdy. When I open a different terminal shell and login I can see that my home directory is not decrypted! When I manually mount my home directory and restart sddm I can login via howdy!

Another test confirms that encrypted home directories are not decrypted when howdy is used:
Instead of sddm I open a second terminal shell with CTRL+ALT+F2 and login with howdy. The login is working but I start in an empty home directory while my home directory is still encrypted.
If I disable howdy and login on the terminal shell the home directory is decrypted and I start inside my home.

So I think when I'm using howdy for login the decryption for the home directory does not get triggered anymore - a wrong configuration with PAM?!

_/etc/pam.d/common-session_
```
session [default=1]                     pam_permit.so
session requisite                       pam_deny.so
session required                        pam_permit.so
session optional                        pam_umask.so
session required        pam_unix.so 
session optional                        pam_sss.so 
session optional        pam_ecryptfs.so unwrap
session optional                        pam_mkhomedir.so 
session optional        pam_systemd.so 
```

_/etc/pam.d/common-auth_
```
auth    [success=4 default=ignore]        pam_python.so /lib/security/howdy/pam.py
auth    [success=3 default=ignore]      pam_fprintd.so max_tries=1 timeout=10 # debug
auth    [success=2 default=ignore]      pam_unix.so nullok_secure try_first_pass
auth    [success=1 default=ignore]      pam_sss.so use_first_pass
auth    requisite                       pam_deny.so
auth    required                        pam_permit.so
auth    required                        pam_group.so use_first_pass
auth    optional        pam_ecryptfs.so unwrap
auth    optional                        pam_cap.so
```

**boltgolt**, do you have any ideas to change configuration for more tests?

boltgolt, 2019-11-11:
To simplify: Automatic homedir decryption is done by taking the password you enter on login and using that as the decryption key. However, if you login with Howdy there is no password (authtoken) entered and the home folder fails to mount. I'm not sure if there's a workaround for this

orclex, 2019-11-13:
Bad luck that it is still not possible to decrypt partitions with biometric data like fingerprint or face detection directly.

However, I did some more tests and find out that SDDM only gets stuck if howdy is used and **recognizes** my face.
If I also enter a wrong password and I'm looking to the side while face detection is running so that it couldn't verify my face SDDM does **not** get stuck and returns with focus in the password input field .
If I enter the correct password and I'm looking to the side while face detection is running **I can login**! 
If I enter the correct password and I'm looking into the IR sensor while face detection is running SDDM gets stuck again and can only be restarted from another terminal.

So there is a problem when howdy recognizes my face and the home directory is encrypted which cannot be decrypted with howdy. 
The better way would be to ignore face detection if home directory _/home/user_ is an encrypted partition and not decrypted. Howdy could return with a message that home directory has to be decrypted by a login with password fist.
If _/home/user_ is an encrypted partition and already decrypted howdy works and should be used for login! This is the case if you logout after first login with a password and want to login again.

This would allow the user to enable howdy in spite of encrypted home directories and use it for unlocking screen or sudo.

To find out if the home directory is encrypted maybe this could help you.
An encrypted and not decrypted home directory contains files like:

```
# ll /home/alex/
insgesamt 8
dr-x------ 2 alex alex 4096 Sep 25 08:56 ./
drwxr-xr-x 6 root root 4096 Okt  7 08:47 ../
lrwxrwxrwx 1 alex alex   56 Sep 25 08:56 Access-Your-Private-Data.desktop -> /usr/share/ecryptfs-utils/ecryptfs-mount-private.desktop*
lrwxrwxrwx 1 alex alex   30 Sep 25 08:55 .ecryptfs -> /home/.ecryptfs/alex/.ecryptfs/
lrwxrwxrwx 1 alex alex   29 Sep 25 08:55 .Private -> /home/.ecryptfs/alex/.Private/
lrwxrwxrwx 1 alex alex   52 Sep 25 08:56 README.txt -> /usr/share/ecryptfs-utils/ecryptfs-mount-private.txt
```
Also there is a directory _/home/.ecryptfs_ for encrypted home directories. In my case I find the target mount point in

```
$ cat /home/.ecryptfs/alex/.ecryptfs/Private.mnt
/home/alex
```

So you could find out if a users home directory is encrypted if there exists a file _/home/.ecryptfs/$USER/.ecryptfs/Private.mnt_ which contians the users home directory _/home/$USER_.
You could grep mount for `/home/.ecryptfs/$USER/.Private on /home/$USER type ecryptfs`
to get if it the home directory is decrypted and face detection can be used.

Is this possible?
Thanks!

boltgolt, 2019-11-13:
I'd go with another route: Disable Howdy on the login screen but enable it in specific other parts, the sudo command for example.

Remove the default Howdy PAM config:

```
rm /usr/share/pam-configs/howdy
pam-auth-update --package
```

Then, add Howdy back to some contexts manually. Try looking at the files in `/etc/pam.d/`, each of which is used for different places to authenticate. To enable Howdy in one if these places, add the following line after the opening comments in the files:

```
auth sufficient pam_python.so /lib/security/howdy/pam.py
```

orclex, 2019-11-13:
The problem is that the main authentication is in _/etc/pam.d/common-auth_ which most of the pam scripts include. After you have disabled this you neet to enable howdy in any other pam file like sudo if the service uses its own pam file.
Unfortunately there is no pam file for the lock screen! If I cannot use howdy for login on encrypted home directories, I'd at least love to use face detection to unlock my locked session which also works but seems to only use the common-auth pam file. So I cannot disable howdy in common-auth.
Additionally all pam files you have changed manually will be overwritten on next package update...
My idea would allow the user to make maximum use of howdy without manual adjustment.

Of course it would be great to choose between different services for authenticating with howdy in the config file and for example disable howdy for sudo but that's another thing...

retomi, 2020-08-06:
> Could you try setting `no_confirmation` to true in the config?

That worked for me, thanks. Kubuntu 20.10

Being-Rts, 2021-03-25:
I mistakenly uninstalled breeze theme in KDE in Kubuntu. 
Now when I am trying to login after entering password nothing is happening. 
Login freeze problem

closed 2021-04-02 by boltgolt