boltgolt/howdy #210: apt-get remove fails due to "FileNotFoundError: [Errno 2] No such file or directory: 'pip3': 'pip3'" when pip3 is installed

krichter722, 2019-05-14:
`sudo apt-get remove howdy` fails due to

```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  fswebcam libpam-python streamer xawtv-plugins
Use 'sudo apt autoremove' to remove them.
The following packages will be REMOVED:
  howdy
0 upgraded, 0 newly installed, 1 to remove and 4 not upgraded.
After this operation, 151 kB disk space will be freed.
Do you want to continue? [J/n] j
(Reading database ... 867643 files and directories currently installed.)
Removing howdy (2.5.1) ...
Can't remove executable
Traceback (most recent call last):
  File "/var/lib/dpkg/info/howdy.prerm", line 45, in <module>
    subprocess.call(['pip3', 'uninstall', 'dlib', '-y', '--no-cache-dir'])
  File "/usr/lib/python3.7/subprocess.py", line 323, in call
    with Popen(*popenargs, **kwargs) as p:
  File "/usr/lib/python3.7/subprocess.py", line 775, in __init__
    restore_signals, start_new_session)
  File "/usr/lib/python3.7/subprocess.py", line 1522, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'pip3': 'pip3'
dpkg: error processing package howdy (--remove):
 installed howdy package pre-removal script subprocess returned error exit status 1
Errors were encountered while processing:
 howdy
libdvd-pkg: Package libdvdcss2-1.4.2-1 was removed, stop processing...
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

whereas `stat $(which pip3)` shows that `pip3` is installed and executable:

```
  File: /usr/local/bin/pip3
  Size: 215       	Blocks: 8          IO Block: 4096   regular file
Device: 802h/2050d	Inode: 1003045     Links: 1
Access: (0755/-rwxr-xr-x)  Uid: (    0/    root)   Gid: (    0/    root)
Access: 2019-05-09 19:59:40.259943550 +0200
Modify: 2019-05-09 19:59:40.255943455 +0200
Change: 2019-05-09 19:59:40.255943455 +0200
 Birth: -
```

It's however not located in `/usr/bin/`, but in an equally realistic location.

----

I've searched for similar issues already, and my issue has not been reported yet. (/)

Linux distribution (if applicable): Ubuntu 19.04

Howdy version: 2.5.1 from `ppa:boltgolt/howdy`

krichter722, 2019-05-14:
I just realized that `python3-pip` which provides a `/usr/bin/pip3` (and is probably a dependency of `howdy`) was installed, however `/usr/bin/pip3` deleted.

closed 2019-05-17 by krichter722

iamzko, 2022-01-25:
My pip3 is in /usr/local/bin.So I try ln -s /usr/local/bin/pip3 /usr/bin/pip3.This problem is solved.Perfect!