boltgolt/howdy #212: Face Detection on login screen

saurabhdhillon, 2019-05-15:
1. 
I am using ubuntu 16.04 LTE, and facing "Unknown error: 1". At login when i press enter with no password, cam turns on and message is displayed "Unknown error: 1".
I have added 7 8 face models for more accuracy.
And in case when i am already logged in and authenticate popup appears, camera turns on and message is displayed "face detection timeout reached"

2. 
Output for : "v4l2-ctl -d /dev/video0 --list-formats"
```
ioctl: VIDIOC_ENUM_FMT
	Index       : 0
	Type        : Video Capture
	Pixel Format: 'MJPG' (compressed)
	Name        : Motion-JPEG

	Index       : 1
	Type        : Video Capture
	Pixel Format: 'YUYV'
	Name        : YUYV 4:2:2
```
No output for second command(dmesg | grep "Unknown video format").

3.
On running **sudo -i**, it detects face without any issue
```
saurabh@DV-LT-W-017113:~$ sudo -i
Identified face as saurabh
```
4.
I also checked **/var/log/auth.log** files after a failed login attempt by howdy, eventually i logged in with password
```
May 14 15:21:35 DV-LT-W-017113 unix_chkpwd[27612]: password check failed for user (saurabh)
May 14 15:21:35 DV-LT-W-017113 compiz[1752]: pam_unix(unity:auth): authentication failure; logname= uid=1000 euid=1000 tty= ruser= rhost=  user=saurabh
May 14 15:21:39 DV-LT-W-017113 unix_chkpwd[27631]: password check failed for user (saurabh)
May 14 15:21:39 DV-LT-W-017113 compiz[1752]: pam_unix(unity:auth): authentication failure; logname= uid=1000 euid=1000 tty= ruser= rhost=  user=saurabh
May 14 15:21:44 DV-LT-W-017113 compiz[1752]: gkr-pam: unlocked login keyring

```

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Ubuntu 16.04 LTS

Howdy version: 2.5.1

saurabhdhillon, 2019-05-16:
@boltgolt Any suggestion on how i can fix this

AnthonyWharton, 2019-05-16:
First of all, 7-8 models should not be necessary and will probably only slow down your system. I run with just 2 models and it works fine in a wide variety of scenarios.

Secondly, your issue is not very clear to me. I can see that facial detection works with `sudo -i`, but you say you're getting an error - is this just on the Ubuntu Login Screen?  This sounds like it could be something to do with PAM, but I'm not that familiar with how Howdy installs itself from the `.deb` package.

Finally, as a general piece of advice for open source projects; software is open and free for you to use, as a result not everyone has time to reply to issues quickly as it's likely they have their own life to live as they are not making a living from these projects and they are a hobby.  Be patient, help will come :) Badgering people is more likely for you to get a terse response or no help at all.

saurabhdhillon, 2019-05-16:
@AnthonyWharton Will keep the last bit in mind. Thanks. Will try coming up with a solution for this. :)

boltgolt, 2019-06-09:
Could you switch to TTY1 (ctrl + alt + F1) on login? Howdy should be able to print its crashlog there. You can switch back to graphical with ctrl + alt + F7

boltgolt, 2020-06-21:
Closing due to inactivity

closed 2020-06-21 by boltgolt