boltgolt/howdy #215: Enabling IR Led Camera

raknerak, 2019-05-23:
I started to use HP Zbook 14u g5 notebook and this notebook comes with windows 10 pro by default and I wiped all disk data and I have installed fresh 18.04 ubuntu on this notebook. This notebook comes with  HD Camera, IR Camera, Webcam LED, IR Camera LEDs  (2) combination and when I was testing face recognition on windows IR Camera LEDs (2) were blinking during face recognition. 

http://h20195.www2.hp.com/v2/getpdf.aspx/c05830243.pdf

After ubuntu installation, I have installed howdy application and Linux Os detected all video hw as 4 different device id. Howdy selected one of them and face detection but it was not working. I have changed howdy config file with trying all device IDs and I suppose I found correct device for IR Camera and howdy started working well. During face recognition Webcam LED is being ON state and IR Camera Leds are not working. It causes time to face detection failures due to environment light conditions. Is it possible to enable using IR Camera LEDs for better detection performance on howdy?

boltgolt, 2019-08-02:
Try using the `sudo howdy test` command to see what's going wrong. If there's not circle around your face then it's not detecting you

boltgolt, 2020-06-21:
Closing due to inactivity

closed 2020-06-21 by boltgolt