boltgolt/howdy #218: Require confirmation of recognition

mayou36, 2019-06-02:
Hi, first of all: many thanks for this little peace of code, great work!

I do currently though have slight security concerns about using it. I do not fear abusive colleagues or me mistyping sudo, but a peace of malware code that tries to execute sudo. Just because I am sitting in front of the laptop does not mean that I want everything that tries to execute with sudo to be executed with sudo.

Is it possible to add a required confirmation before the face recognition is successful? This can be as little as pressing "Enter". So the face get's recognized and then "Enter" has to be pressed to proceed.

This would allow to be sure about what actually uses sudo and what should not.

boltgolt, 2019-06-09:
Good point, not sure how to implement this with PAM though. Needs be be looked into

stinobook, 2019-06-11:
i think a good solution to this might be to make a 2nd 'sudo' command which calls a script that adds howdy to the sudo pam file and removes it afterwards. 
for security reasons i think it would be best to let the user decide an alias to do this so the malware cannot just use this command instead.

lennart-k, 2019-07-02:
As far as I can see it seems fairly easy to just wait for a keypress with https://python-libinput.readthedocs.io/en/latest/index.html
The problem would just be that this keypress could also affect the program you are currently using.

boltgolt, 2019-08-02:
Additionally, that would only work in the console and would hang authentication forever on the login or lock screen.

boltgolt, 2021-04-02:
The concept of rubber stamps is being implemented in 3.0.0

boltgolt, 2026-11-16:
Hey, thanks for suggesting this! This is now possible in v3 with rubber stamps. Set `enabled = true` under `[rubberstamps]` in the config and add a rule like `hotkey 5s failsafe`. After your face is recognised, Howdy waits for you to press Enter to authorise (or Esc to abort) and cancels if you don't confirm within 5 seconds. Closing this as implemented!

closed 2026-11-16 by boltgolt