boltgolt/howdy #219: not activating in kde lock screen

ShaunJorstad, 2019-06-06:
I can't get howdy to activate on the KDE plasma lock screen. Both the SDDM login screen and the terminal launch howdy correctly. 

I have added  `auth sufficient pam_python.so /lib/security/howdy/pam.py` to `/etc/pam.d/kde`. I also added the same line to system-auth, system-local-login, and system-login, none of which changed any functionality. 

I don't believe any errors are being logged, but I may not be looking for them in the correct locations.

----

I looked through the arch wiki and former bug reports and didn't see my issue previously reported but I could be blind. 

Linux distribution (if applicable): 
Manjaro Linux
Kernel: 4.19.45-1-MANJARO

Howdy version: 2.5.0-1

yashahir, 2019-06-06:
Same problem for me as well. Does it work for you in anything else other than sudo? I've tried launching applications which require root password (like KDE Partition) and settings which only root can change like the SDDM settings. Nothing else works other than sudo. Can somebody help with this?

ShaunJorstad, 2019-06-07:
I haven't encountered any other instance where it doesn't work. Both KDE partition and applying a change to SDDM settings activated the IR cameras and authenticated me. However both of these cause an infinite loop where my ir scanners activate and appear to scan my face every couple seconds. I have found the only way to terminate this loop is through a reboot. (I'm probably just not killing the proper service)

yashahir, 2019-06-07:
Wait, what files did you put the `auth sufficient pam_python.so /lib/security/howdy/pam.py to /etc/pam.d/kde` line in? I'll try to follow you and replicate your issue if possible.

yashahir, 2019-06-07:
Does SDDM work perfectly for you for face unlock? Also what files did you edit to make SDDM work? I wouldlike to know since for me howdy dosen't seem to work with either SDDM or any other KDE app or the lockscreen. The only thing that works perfectly is sudo.

yashahir, 2019-06-07:
Alright so, I tested a few times and apparently, after you insert `auth sufficient pam_python.so /lib/security/howdy/pam.py` to `/etc/pam.d/kde` and `/etc/pam.d/system-auth` makes everything work. SDDM works but you have to at least press enter so that it throws a incorrect password error and then when the field is available again, howdy starts scanning which does unlock the screen. Also, by default the login screen does not work. But if you say open something like KDE partition and when the IR cameras go in the infinite loop, locking your screen will unlock it if the IR cameras scan it. So it does work but only when it's actively being scanned. Meaning adding some sort of trigger to the pam config (I'm not knowledgeable enough) so that when you get to the password field, you can activate the cameras will work.

ShaunJorstad, 2019-06-07:
I believe we are experiencing the same results now. 

I have been unable to get the kde lock screen to trigger howdy/the ir cameras. When stuck in the infinite loop, howdy will authenticate me from the lock screen as well.

It would also be convenient to have the ir cameras activate as soon as the computer is awakened from a suspended state (so as to trigger howdy upon opening a laptop lid as opposed to having to hit enter in the password field first.)

yashahir, 2019-06-07:
It seems the problem might be with sddm and kde at the end of the day since I've read quite a few posts about problems with any kind of biometric authentication...

ShaunJorstad, 2019-06-09:
Is there a way that we can trigger howdy to activate from a script? 

when resuming from a suspended state systemctl executes every script located in `/usr/lib/systemd/system-sleep` with the first argument being 'post'

```bash
#!/bin/bash

if [ $1 == "post" ]
	then
	{script to trigger howdy could go here}	
fi
```

stinobook, 2019-06-11:
Hi all,

i too was trying to figure out something and when trying to replace kde lock with i3lock, and getting i3lock to work. somehow it also fixed my unlocking with kde plasma lock screen.

the simple solution was:
sudo chmod -R 755 /lib/security/howdy

i had noticed in my logs that somehow it executed as a user and not as root.
this also fixes howdy working for i3lock and sddm for me.

i prefer system-auth not having the howdy login so this is a better solution to me

ShaunJorstad, 2019-06-14:
> sudo chmod -R 755 /lib/security/howdy
This allowed the kde lock screen to initiate howdy for me as well. The only issue is that it initiates the same infinite loop as other applications, like the KDE Partition Manager.

 Does i3lock initiate howdy immediately for you, or do you have to hit enter in the password field first, (similar to sddm and kde)?

stinobook, 2019-06-14:
> > sudo chmod -R 755 /lib/security/howdy
> 
> This allowed the kde lock screen to initiate howdy for me as well. The only issue is that it initiates the same infinite loop as other applications, like the KDE Partition Manager.

Infinite loop? if you mean that after unlocking howdy keeps trying to do something, it has something to do with this:

```
# Auto dismiss lock screen on confirmation
# Will run loginctl unlock-sessions after every auth
# Experimental, can behave incorrectly on some systems
dismiss_lockscreen = false
```
after unlocking it keeps trying to do this command even though i've set it to false

> 
> Does i3lock initiate howdy immediately for you, or do you have to hit enter in the password field first, (similar to sddm and kde)?

i have to press enter of course,
since there is no difference between a lock from AFK or a lock from wake, 

if you wouldnt have to press enter then it would keep trying to unlock when you're afk.

maybe there is a way to also issue an unlock when moving the mouse.

yashahir, 2019-06-14:
Did you modify any files other than `/etc/pam.d/kde` and `/etc/pam.d/system-auth`? @stinobook

stinobook, 2019-06-14:
> Did you modify any files other than `/etc/pam.d/kde` and `/etc/pam.d/system-auth`? @stinobook

i do not have it in system-auth :) see below where i have added it.

```
╰─ sudo ack howdy      

sddm
2:auth sufficient pam_python.so /lib/security/howdy/pam.py

i3lock
5:auth sufficient pam_python.so /lib/security/howdy/pam.py

kde
2:auth sufficient pam_python.so /lib/security/howdy/pam.py

```

yashahir, 2019-06-14:
I see. So then, only two issues remain then?
1) Fix infinite looping IR cameras.
2} Make howdy work when moving the mouse on lock screen/getting to sddm without pressing enter.

stinobook, 2019-06-14:
> I see. So then, only two issues remain then?
> 
>     1. Fix infinite looping IR cameras.
> 
>     2. Make howdy work when moving the mouse on lock screen/getting to sddm without pressing enter.

well, for 2 i think its more a locker-issue and not howdy issue. they need to add a check if howdy is installed and if it is issue an unlock when moving the mouse. i doubt we can do anything else about it since its the locker that calls pam which calls howdy.

ShaunJorstad, 2019-06-27:
Just came across #139 
Changing my config to exclude `false` resolved the looping ir cameras.
> ```
> # Auto dismiss lock screen on confirmation
> # Will run loginctl unlock-sessions after every auth
> # Experimental, can behave incorrectly on some systems
> dismiss_lockscreen =
> ```

closed 2019-06-27 by ShaunJorstad

yashahir, 2019-06-27:
Wait what file do you have to edit for this?

ShaunJorstad, 2019-06-27:
config.ini
This was stored at `/usr/lib/security/howdy` for me.
`sudo howdy config`  should open the file with nano. 
@yashahir

yashahir, 2019-06-27:
Thanks, I'll try it!

sbespalov, 2020-08-17:
still not fixed in `Howdy 2.6.0`
following helped from above:
```
$sudo chmod -R 755 /lib/security/howdy
```
> insert `auth sufficient pam_python.so /lib/security/howdy/pam.py` to `/etc/pam.d/kde` and `/etc/pam.d/system-auth`

4fthawaiian, 2022-01-07:
> the simple solution was: sudo chmod -R 755 /lib/security/howdy

this right here fixed it for me in xfce4. cheers!