boltgolt/howdy #220: detect face with rotated camera

berndso, 2019-06-08:
I have howdy running on a Thinkpad X1 Tablet gen3 and it works beautifully in general.
However, faces are only detected when the tablet/camera is in its standard landscape orientation. Adding a model or running howdy test fails in portrait mode. 
It would be great to rotate the camera input before the face recognition algorithm kicks in, to make sure it matches the physical orientation. Maybe this can be achieved by checking the position sensors or even better,  the user interface orientation?

boltgolt, 2019-06-09:
Although it sounds reasonable, i think it's out of scope for the project and would only be used by a very small subset of Howdy users. I'm closing this for now to focus development elsewhere.

If anyone would also like to see this implemented, feel free to add a comment here or propose a PR.

closed 2019-06-09 by boltgolt

denics, 2020-03-30:
I am trying to implement a script that, when rotation is detected, instead of launching howdy, a howdy wrapper is launched that does, more or less:
- modprobe v4l2loopback (this is mandatory, it should be decided how to load it)
- catch the dummy device and launch
    ``ffmpeg -f v4l2 -i /dev/video0 -vf "vflip" -f v4l2 /dev/dummy``
- change *device_path* in ``/lib/secure/howdy/config.ini`` to ``/dev/dummy``

doing this manually it works, maybe there is some more elegant way to achieve the same results?

Best,
Denis

denics, 2020-04-06:
hi @boltgolt do you have an idea if this is "scriptable"? I do not know the security system enough to see if what I am saying would be just a waste of time

memojo, 2023-12-22:
It would be great to have this feature out of the box with howdy! I also have a convertible and howdy does not work in tent mode. One idea for a workaround: just train additional faces in howdy when the camera is flipped.

memojo, 2023-12-22:
Training additional faces in flipped mode works, i just tried it :)