boltgolt/howdy #222: Disable howdy with external monitor

stinobook, 2019-06-18:
Hello,

would there be a possibility to detect an external monitor and disable howdy if it is connected to one ?

In the config i found:
```
# Disable howdy in the PAM
# The howdy command will still function
disabled = false
```
so it would just have to be a switch on this option that works when connecting/disconnecting an external monitor.

the reason being that when im connected to a dock and using desktop peripherals the laptop is closed. when i do anything where howdy is activated i first have to wait for the timeout before i get a password request due to a timeout.

thanks in advance,

Stijn

boltgolt, 2019-06-18:
Howdy has an option to ignore PAM calls when a laptop lid is closed, is your `ignore_closed_lid` config value set to true?

stinobook, 2019-06-18:
> Howdy has an option to ignore PAM calls when a laptop lid is closed, is your `ignore_closed_lid` config value set to true?

my mistake!

since i installed howdy via AUR we apparantly still get the previous version, and i see now that you added this option in the latest release.

Closing thread then! :) and gonna try to upgrade howdy.

thanks mate!

closed 2019-06-18 by stinobook

stinobook, 2019-06-18:
Hello,

i have updated the pkgbuild for arch so it would install the latest version.

since i dont know much about github or pkgbuilds i apparantly fork/cloned your repository.

see here for the update/patch:

https://github.com/stinobook/howdy/commit/9c7977e3d8fe2ff0c253e6b4f58eca15150fb71e